openwrt/asu #158: opkg is installed automatically

McChary, 2019-02-23:
I always try the latest compilations of openwrt and remove many packages that I do not use to accelerate the device, but when I remove opkg from the list it seems to be installed automatically.

You can see the created image.
https://chef.libremesh.org/json/openwrt/snapshots/ar71xx/tiny/tl-wr841-v8/5017f55374dcd40/buildlog-16d10b5f1c774e8.txt

aparcar, 2019-03-06:
I tried the same [here](https://chef.libremesh.org/#34633af51b05) and it works as excepted. Please send the list of packages used in the chef frontend.

McChary, 2019-03-07:
Well apparently in version 18.06.2 you can remove opkg, but in the snapshot versions the first time you open the thread was installed automatically, I just tried and now a construction error is coming out.

https://chef.libremesh.org/download/faillogs/faillog-9e251fa74e79.txt

[image]

Package list:
ip6tables
odhcp6c
base-files
mtd
fstools
busybox
wpad-mini
kmod-gpio-button-hotplug
logd
swconfig
kmod-ath9k
dnsmasq
dropbear
ppp
netifd
ppp-mod-pppoe
uci
libc
uclient-fetch
uboot-envtools
kmod-ipt-offload
libgcc
odhcpd-ipv6only
iptables
firewall
luci

aparcar, 2020-02-25:
opkg removal seem to work when you specify the package you want to remove, see screenshot. Please test the new server at https://chef.libremesh.org/

[image]

aparcar, 2020-02-25:
It uses a different device but this is due to an upstream problem breaking the ath79 target.

closed 2020-02-25 by aparcar