openwrt/asu #165: linksys-wrt32x

Roland-B, 2019-03-09:
Hi, Support for the router linksys-wrt32x seems to have dissapeared from ImageBuilder normal and beta? Is it coming back soon? Thanks.

aparcar, 2019-03-09:
possible that the target isn't activated. which one is it?

dangowrt, 2019-03-09:
`mvevu/cortexa9`

Roland-B, 2019-03-09:
> 
> 
> `mvebu/cortexa9`

beat me to it. Yup, mvebu/cortexa9, cheers for the quick reply.

Roland-B, 2019-03-10:
Didnt mean to close, appologies. Issue still exists.

aparcar, 2019-03-10:
hey, there are typos in the config file.

    - mvevu/cortexa53
    - mvevu/cortexa72
    - mvevu/cortexa9

 please correct it via a PR in ./asu/utils/distributions/openwrt/distro_config.yml

Roland-B, 2019-03-10:
Thanks apacar, but im not running my own server im using the online ones below which are currenlty failing due to the issue reported.

https://as-test.stephen304.com/
https://chef.libremesh.org/
https://betachef.libremesh.org/

TommyMacMoo, 2019-03-11:
I, sadly have same issue, I WRT 32X and also WRt 3200 dissapeared ...will wait...and see..if get fixed

commit 33290270 (2019-03-11): Merge pull request #166 from dangowrt/master

distributions: fix typo in targets
fixes #165
closed 2019-03-11 by dangowrt