openwrt/asu #169: Requests from OpenWrt don't work (While on the website they work)

tomjschwanke, 2019-03-17:
OpenWrt 18.06.2 on TL-WDR4300 V1
The luci-app finds possible package upgrades, but clicking on "Request Firmware" just greys out the button and I'm not getting a firmware (I have waited long, even my login timed out).

Builds requested on the Website work fine though.

aparcar, 2019-03-18:
should be fixed, please try again

aparcar, 2019-03-18:
https://github.com/aparcar/attendedsysupgrade-server/commit/2c485fc645981dd9a2c5c7c6b77e93e7a28cfd41

tomjschwanke, 2019-03-18:
Tried it again today. It got to "processing request" which also seemed to load forever, so I refreshed the interface and its back to the above problem.

aparcar, 2019-03-18:
I see, the request fails and the error response is not (yet) handled by the luci-app... I'm on it

Roland-B, 2019-04-05:
Same here for WRT32X, but also get Unsupported device, release, target, **_subtraget_** or board  <-- note spelling mistake also for the error.

aparcar, 2020-03-07:
I reworked the app and web frontend and now everything should work again. If not please reopen

closed 2020-03-07 by aparcar