openwrt/asu #170: TL-WDR4300 v1/OpenWrt 18.06.2: Wrong version of luci-ssl and luci-app-* in image

rkunze, 2019-03-31:
I'm trying to upgrade a TL-WDR4300 v1 with OpenWrt 18.06.2 with this package list:
```
bash
busybox
luci-app-commands
fstools
coreutils
wpad-mini
ip6tables
unbound
odhcp6c
uci
socat
logd
ip-full
coreutils-sleep
libc
netifd
kmod-usb-core
dropbear
kmod-usb-hid
picocom
kmod-usb-serial-pl2303
opkg
uboot-envtools
kmod-gpio-button-hotplug
tcpdump
luci-app-attendedsysupgrade
kmod-usb-serial
luci-app-unbound
luci-ssl
uclient-fetch
kernel
iptables
unbound-control
odhcpd-ipv6only
ppp-mod-pppoe
kmod-usb-ledtrig-usbport
kmod-ath9k
sispmctl
triggerhappy
curl
base-files
openssh-sftp-server
coreutils-stty
kmod-usb2
swconfig
less
dnsmasq-full
mtd
evtest
firewall
ppp
```

Attended-sysupgrades shows these updates:
```
luci-app-commands: git-19.051.55698-76cf653-1 to git-19.079.57770-b99e77d-1
luci-app-unbound: git-19.051.55698-76cf653-1 to git-19.079.57770-b99e77d-1
luci-app-attendedsysupgrade: git-19.051.55698-76cf653-1 to git-19.079.57770-b99e77d-1
luci-ssl: git-19.051.55698-76cf653-1 to git-19.079.57770-b99e77d-1
```

`Request firmware` results in
```
Firmware created: openwrt-18.06.2-4c06e5273fc0f81-ar71xx-generic-tl-wdr4300-v1-squashfs-sysupgrade.bin Build log
```
Image link: https://chef.libremesh.org/json/openwrt/18.06.2/ar71xx/generic/tl-wdr4300-v1/4c06e5273fc0f81/openwrt-18.06.2-4c06e5273fc0f81-ar71xx-generic-tl-wdr4300-v1-squashfs-sysupgrade.bin
Build log: https://chef.libremesh.org/download/openwrt/18.06.2/ar71xx/generic/tl-wdr4300-v1/4c06e5273fc0f81/buildlog-5ceb2c46bf33a0c.txt

But the version of the luci packages in this image is still at git-19.051* (see the build log).

Note: On a TL-WR841N v9, an similar upgrade (luci-ssl and some luci modules, from 19.051* to 19.079*) worked without any problems

aparcar, 2019-04-01:
Hi @rkunze created images are cached and your build request points to an outdated image. Obviously the outdated image should be removed automatically, however this functionality was temporary removed due to it's "cost". I'm working on faster and less database intense version, however not yet finished.
The rough idea is like [this](https://dpaste.de/mHpj), meaning to scan daily for images containing outdated version. I'll let you know once this is fixed.

aparcar, 2019-04-06:
@dangowrt can you help me out here?

aparcar, 2019-04-08:
I added a workaround by deleting all images older than 14 days.

rkunze, 2019-04-10:
The workaround worked as expected - thanks!

aparcar, 2019-04-11:
I'll close this issue for now, however will continue with a database improvement

closed 2019-04-11 by aparcar