openwrt/asu #172: Building snapshot image fails

rkunze, 2019-05-28:
I am trying to build an standard OpenWRT snapshot image (flavor "OpenWRT default packages only", but other variants failed as well with the same error) for "Ubiquiti EdgeRouter X". 
Build fails with error message "Incompatible package selection", the [corresponding build log](https://chef.libremesh.org/download/faillogs/faillog-bf5ac52ae0ab.txt) shows these errors:

```
Collected errors:
 * opkg_install_pkg: Package size mismatch: swconfig is 9408 bytes, expecting 9410 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7433 bytes, expecting 7424 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: dropbear is 86335 bytes, expecting 86308 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: uci is 7185 bytes, expecting 7180 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: libubox20170601 is 19250 bytes, expecting 19247 bytes
 * opkg_install_cmd: Cannot install package fstools.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 23884 bytes, expecting 23890 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: opkg is 58559 bytes, expecting 58548 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: libuci20130104 is 15454 bytes, expecting 15463 bytes
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10213 bytes, expecting 10218 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: libubox20170601 is 19250 bytes, expecting 19247 bytes
 * opkg_install_cmd: Cannot install package mtd.
 * opkg_install_pkg: Package size mismatch: logd is 10177 bytes, expecting 10175 bytes
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_pkg: Package size mismatch: ppp is 97872 bytes, expecting 97871 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 30124 bytes, expecting 30153 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 101940 bytes, expecting 101947 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: firewall is 40969 bytes, expecting 40976 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 101940 bytes, expecting 101947 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: dropbear is 86335 bytes, expecting 86308 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: firewall is 40969 bytes, expecting 40976 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: logd is 10177 bytes, expecting 10175 bytes
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 23884 bytes, expecting 23890 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 30124 bytes, expecting 30153 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 58559 bytes, expecting 58548 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: ppp is 97872 bytes, expecting 97871 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10213 bytes, expecting 10218 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: swconfig is 9408 bytes, expecting 9410 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7185 bytes, expecting 7180 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7433 bytes, expecting 7424 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
```

Constm, 2019-06-10:
Similar complain on package size mismatch with attempt to build for different router. Collected errors:
 * opkg_install_pkg: Package size mismatch: libnl-tiny is 13165 bytes, expecting 13167 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
 * opkg_install_pkg: Package size mismatch: swconfig is 9474 bytes, expecting 9469 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: libubox20170601 is 18453 bytes, expecting 18457 bytes
 * opkg_install_cmd: Cannot install package luci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7692 bytes, expecting 7696 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: dropbear is 83576 bytes, expecting 83575 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: uci is 7497 bytes, expecting 7504 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: libubox20170601 is 18453 bytes, expecting 18457 bytes
 * opkg_install_cmd: Cannot install package fstools.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 26031 bytes, expecting 26019 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: opkg is 66851 bytes, expecting 66856 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: libuci20130104 is 15570 bytes, expecting 15542 bytes
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Package size mismatch: ath10k-firmware-qca4019-ct is 459061 bytes, expecting 459054 bytes
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct.
 * opkg_install_pkg: Package size mismatch: libnl-tiny is 13165 bytes, expecting 13167 bytes
 * opkg_install_cmd: Cannot install package kmod-ath10k-ct.
 * opkg_install_pkg: Package size mismatch: libubox20170601 is 18453 bytes, expecting 18457 bytes
 * opkg_install_cmd: Cannot install package mtd.
 * opkg_install_pkg: Package size mismatch: logd is 10612 bytes, expecting 10619 bytes
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_pkg: Package size mismatch: netifd is 69386 bytes, expecting 69375 bytes
 * opkg_install_cmd: Cannot install package netifd.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 34452 bytes, expecting 34451 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 115521 bytes, expecting 115526 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: firewall is 46908 bytes, expecting 46909 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: ath10k-firmware-qca4019-ct is 459061 bytes, expecting 459054 bytes
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 115521 bytes, expecting 115526 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: dropbear is 83576 bytes, expecting 83575 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: firewall is 46908 bytes, expecting 46909 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: logd is 10612 bytes, expecting 10619 bytes
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_pkg: Package size mismatch: netifd is 69386 bytes, expecting 69375 bytes
 * opkg_install_cmd: Cannot install package netifd.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 26031 bytes, expecting 26019 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 34452 bytes, expecting 34451 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 66851 bytes, expecting 66856 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: swconfig is 9474 bytes, expecting 9469 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7497 bytes, expecting 7504 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7692 bytes, expecting 7696 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

antoan-boss, 2019-06-18:
Same size mismatch errors. Snapshot/ZBT WE-1326/with web interface also without. I tried 18.06.2 and it builds w/o errors.

Collected errors:
 * opkg_install_pkg: Package size mismatch: libnl-tiny is 12841 bytes, expecting 12833 bytes
 * opkg_install_cmd: Cannot install package kmod-mt76x2.
 * opkg_install_pkg: Package size mismatch: libnl-tiny is 12841 bytes, expecting 12833 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
 * opkg_install_pkg: Package size mismatch: swconfig is 9403 bytes, expecting 9405 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7432 bytes, expecting 7429 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: dropbear is 86320 bytes, expecting 86329 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: uci is 7182 bytes, expecting 7180 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: libjson-c2 is 14815 bytes, expecting 14813 bytes
 * opkg_install_cmd: Cannot install package fstools.
 * opkg_install_pkg: Package size mismatch: libnl-tiny is 12841 bytes, expecting 12833 bytes
 * opkg_install_cmd: Cannot install package iwinfo.
 * opkg_install_pkg: Package size mismatch: busybox is 209008 bytes, expecting 208963 bytes
 * opkg_install_cmd: Cannot install package busybox.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 23884 bytes, expecting 23888 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: opkg is 58621 bytes, expecting 58612 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: libuci20130104 is 15460 bytes, expecting 15464 bytes
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10296 bytes, expecting 10295 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: libubus20170705 is 10012 bytes, expecting 10007 bytes
 * opkg_install_cmd: Cannot install package logd.
 * opkg_install_pkg: Package size mismatch: ppp is 97925 bytes, expecting 97934 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 30133 bytes, expecting 30125 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 102122 bytes, expecting 102108 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: firewall is 40965 bytes, expecting 40970 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: busybox is 209008 bytes, expecting 208963 bytes
 * opkg_install_cmd: Cannot install package busybox.
 * opkg_install_pkg: Package size mismatch: dnsmasq is 102122 bytes, expecting 102108 bytes
 * opkg_install_cmd: Cannot install package dnsmasq.
 * opkg_install_pkg: Package size mismatch: dropbear is 86320 bytes, expecting 86329 bytes
 * opkg_install_cmd: Cannot install package dropbear.
 * opkg_install_pkg: Package size mismatch: firewall is 40965 bytes, expecting 40970 bytes
 * opkg_install_cmd: Cannot install package firewall.
 * opkg_install_pkg: Package size mismatch: odhcp6c is 23884 bytes, expecting 23888 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 30133 bytes, expecting 30125 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 58621 bytes, expecting 58612 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: ppp is 97925 bytes, expecting 97934 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10296 bytes, expecting 10295 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: swconfig is 9403 bytes, expecting 9405 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7182 bytes, expecting 7180 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7432 bytes, expecting 7429 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

aparcar, 2019-07-07:
Please try the same as described [here](https://github.com/aparcar/attendedsysupgrade-server/issues/175#issuecomment-509030869)

aparcar, 2019-08-14:
Please retry via https://aparcar.stephen304.com/

closed 2019-08-14 by aparcar

jchulce, 2019-08-16:
Having the size mismatch errors again on https://aparcar.stephen304.com/ today
[faillog-e8c230f20372.txt](https://github.com/aparcar/attendedsysupgrade-server/files/3510503/faillog-e8c230f20372.txt)