openwrt/asu #173: pyodbc.InterfaceError while docker-compose up

mpruessmeier, 2019-06-08:
I get a pyodbc.InterfaceError on server_1 while command: docker-compose up

Docker version 18.09.5, build e8ff056
Docker-compose version 1.17.1

`db_1      | 2019-06-08 22:40:42.441 UTC [1] LOG:  listening on IPv4 address "0.0.0.0", port 5432
db_1      | 2019-06-08 22:40:42.441 UTC [1] LOG:  listening on IPv6 address "::", port 5432
db_1      | 2019-06-08 22:40:42.457 UTC [1] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
db_1      | 2019-06-08 22:40:42.483 UTC [71] LOG:  database system was shut down at 2019-06-08 22:40:42 UTC
db_1      | 2019-06-08 22:40:42.493 UTC [1] LOG:  database system is ready to accept connections
db_1      | 2019-06-08 22:40:42.594 UTC [78] LOG:  incomplete startup packet
server_1  | INFO:asu.utils.database:log initialized
server_1  | INFO:asu.utils.database:config initialized
server_1  | Traceback (most recent call last):
server_1  |   File "/usr/local/bin/flask", line 11, in <module>
server_1  |     sys.exit(main())
server_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/cli.py", line 906, in main
server_1  |     cli.main(args=args, prog_name=name)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/cli.py", line 569, in main
server_1  |     return super(FlaskGroup, self).main(*args, **kwargs)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 717, in main
server_1  |     rv = self.invoke(ctx)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1132, in invoke
server_1  |     cmd_name, cmd, args = self.resolve_command(ctx, args)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1171, in resolve_command
server_1  |     cmd = self.get_command(ctx, cmd_name)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/cli.py", line 526, in get_command
server_1  |     rv = info.load_app().cli.get_command(ctx, name)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/cli.py", line 381, in load_app
server_1  |     app = locate_app(self, import_name, name)
server_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/cli.py", line 236, in locate_app
server_1  |     __import__(module_name)
server_1  |   File "/asu/asu/__init__.py", line 8, in <module>
server_1  |     import asu.views
server_1  |   File "/asu/asu/views.py", line 20, in <module>
server_1  |     database = Database(config)
server_1  |   File "/asu/asu/utils/database.py", line 14, in __init__
server_1  |     self.connect()
server_1  |   File "/asu/asu/utils/database.py", line 18, in connect
server_1  |     self.cnxn = pyodbc.connect(connection_string)
server_1  | pyodbc.InterfaceError: ('IM002', '[IM002] [unixODBC][Driver Manager]Data source name not found, and no default driver specified (0) (SQLDriverConnect)')
attendedsysupgradeserver_server_1 exited with code 1
`

aparcar, 2019-08-17:
Thanks for reporting, I backported some fixes here https://github.com/aparcar/attendedsysupgrade-server/commit/86f7215cf450ff8e6b2bf749ea0d649952b5d01b

aparcar, 2020-02-25:
pyodbc is no longer required to run the server.

closed 2020-02-25 by aparcar