openwrt/asu #174: Request firmware returns old firmware

N1ghtshad0w, 2019-06-21:
Search for upgrades returns with Package upgrades available
Request firmware returns with Firmware created
The problem is the link for the created firmware is for the existing firmware (old) not the one with the package upgrades (new)

aparcar, 2019-07-04:
Currently build images of "stable" releases are stored for two weeks, I have to implement a way to automatically remove images which contain outdated packages, but that's not that easy.

I'll try to fix it at some point!

aparcar, 2019-08-14:
What version/release are you building for?

N1ghtshad0w, 2019-08-15:
> 
> 
> What version/release are you building for?

18.06.04

aparcar, 2020-02-25:
The current server at https://chef.libremesh.org/ only supports snapshot and 18.06.x can"t be supported due to missing JSON files.

closed 2020-02-25 by aparcar