openwrt/asu #175: Build for Archer c7 v5 gets error on snapshot

3lk0k0, 2019-06-24:
### BUILD COMMAND:

DEFAULTS_HASH=""
PACKAGES="wpad-basic swconfig kmod-usb2 luci uboot-envtools kmod-gpio-button-hotplug -libgcc uclient-fetch dropbear uci fstools kmod-usb-core kmod-usb-ledtrig-usbport iptables busybox odhcp6c ath10k-firmware-qca988x-ct kmod-ath9k opkg ip6tables base-files -libc ppp-mod-pppoe kmod-ath10k-ct mtd kmod-ipt-offload logd netifd ppp odhcpd-ipv6only dnsmasq firewall"
REQUEST_HASH="33f6822c71b6"
PROFILE="archer-c7-v5"
TARGET="ar71xx/generic"
VERSION="snapshot"
DISTRO="openwrt"
REQUEST_ID="6093"
PACKAGES_HASH="58dd9c881e49"
IMAGE_HASH="None"
./meta image

### STDERR:

gpg: key CD84BCED626471F1: "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg: key 833C6010D52BBB6B: "LEDE Release Builder (17.01 "Reboot" Signing Key) <lede-dev@lists.lede-project.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg: key 0F20257417E1CE16: "OpenWrt Release Builder (18.06 Signing Key) <openwrt-devel@lists.openwrt.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
download checksums and signature
--2019-06-23 18:44:18--  https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/sha256sums
Resolving downloads.openwrt.org (downloads.openwrt.org)... 176.9.48.73, 2a01:4f8:150:6449::2
Connecting to downloads.openwrt.org (downloads.openwrt.org)|176.9.48.73|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 149801 (146K) [text/plain]
Saving to: â€˜sha256sumsâ€™

     0K .......... .......... .......... .......... .......... 34%  926K 0s
    50K .......... .......... .......... .......... .......... 68%  933K 0s
   100K .......... .......... .......... .......... ......    100% 73.9M=0.1s

2019-06-23 18:44:19 (1.32 MB/s) - â€˜sha256sumsâ€™ saved [149801/149801]

--2019-06-23 18:44:19--  https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/sha256sums.gpg
Resolving downloads.openwrt.org (downloads.openwrt.org)... 176.9.48.73, 2a01:4f8:150:6449::2
Connecting to downloads.openwrt.org (downloads.openwrt.org)|176.9.48.73|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/sha256sums.asc [following]
--2019-06-23 18:44:19--  https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/sha256sums.asc
Reusing existing connection to downloads.openwrt.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 837 [text/plain]
Saving to: â€˜sha256sums.ascâ€™

     0K                                                       100% 15.6M=0s

2019-06-23 18:44:19 (15.6 MB/s) - â€˜sha256sums.ascâ€™ saved [837/837]

verifying sha256sums signature
gpg: assuming signed data in 'sha256sums'
gpg: Signature made Mon 17 Jun 2019 05:18:15 AM PDT
gpg:                using RSA key F93525A88B699029
gpg: Good signature from "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" [ultimate]
verified sha256sums signature.
ImageBuilder is up-to-date
Downloading http://downloads.openwrt.org/snapshots/targets/ar71xx/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/../../../../../../../../../../../../home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/dl/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/../../../../../../../../../../../../home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/dl/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/../../../../../../../../../../../../home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/dl/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/../../../../../../../../../../../../home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ar71xx/generic/dl/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.gz
[... 62 lines cut ...]
 * opkg_install_pkg: Package size mismatch: odhcp6c is 23780 bytes, expecting 23779 bytes
 * opkg_install_cmd: Cannot install package odhcp6c.
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 29800 bytes, expecting 29799 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 59018 bytes, expecting 59012 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: ppp is 98391 bytes, expecting 98397 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10323 bytes, expecting 10327 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: swconfig is 9392 bytes, expecting 9398 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7190 bytes, expecting 7193 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 7386 bytes, expecting 7390 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: urandom-seed is 1538 bytes, expecting 1539 bytes
 * opkg_install_cmd: Cannot install package urandom-seed.
 * opkg_install_pkg: Package size mismatch: urngd is 5521 bytes, expecting 5516 bytes
 * opkg_install_cmd: Cannot install package urngd.
 * opkg_install_pkg: Package size mismatch: wpad-basic is 288613 bytes, expecting 288610 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

3lk0k0, 2019-07-06:
im doing something wrong ? thank you

aparcar, 2019-07-07:
Please try to run the build manually. If that fails too, it's not the fault of the server. 

I tried the following:

```
git clone git@github.com:aparcar/meta-imagebuilder.git
cd meta-imagebuilder/ # the server uses the very same script

export DEFAULTS_HASH=""
export PACKAGES="wpad-basic swconfig kmod-usb2 luci uboot-envtools kmod-gpio-button-hotplug -libgcc uclient-fetch dropbear uci fstools kmod-usb-core kmod-usb-ledtrig-usbport iptables busybox odhcp6c ath10k-firmware-qca988x-ct kmod-ath9k opkg ip6tables base-files -libc ppp-mod-pppoe kmod-ath10k-ct mtd kmod-ipt-offload logd netifd ppp odhcpd-ipv6only dnsmasq firewall"
export REQUEST_HASH="33f6822c71b6"
export PROFILE="archer-c7-v5"
export TARGET="ar71xx/generic"
export VERSION="snapshot"
export DISTRO="openwrt"
export REQUEST_ID="6093"
export PACKAGES_HASH="58dd9c881e49"
export IMAGE_HASH="None"
./meta image
```

However, it failed due to some snapshot problems (which are out of my hands):

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	libip4tc2
 * 	libip6tc2
 * opkg_install_cmd: Cannot install package luci.
make[2]: *** [Makefile:156: package_install] Error 255
make[1]: *** [Makefile:117: _call_image] Error 2
make: *** [Makefile:196: image] Fehler 2
```

3lk0k0, 2019-07-07:
Thank you !, seems to be a problem with the snapshot. y flash the build from the web to the router, then i add all the pkg. but i cannot build from the scracht

aparcar, 2019-07-08:
Please close the issue if it works now for you. Also feel free to investigate on why the snapshot fails and report stuff upstream

closed 2019-08-14 by aparcar