openwrt/asu #177: Linksys WRT1900ACS "Unsupported device, release, target, subtraget or board"

apcimino, 2019-07-03:
Hi,

I'd send along the error log but I don't know where that lives.

Cheers,
Anthony

aparcar, 2019-07-07:
Hi, do you use the Luci app?

apcimino, 2019-07-07:
Yes

aparcar, 2019-08-14:
Please post the JSON data you throw again the server

apcimino, 2019-08-14:
{"packages": ["base-files", "busybox", "dropbear", "mtd", "uci", "opkg", "netifd", "fstools", "uclient-fetch", "logd", "swconfig", "wpad-mini", "dnsmasq", "iptables", "ip6tables", "ppp", "ppp-mod-pppoe", "firewall", "odhcpd-ipv6only", "odhcp6c", "iwinfo", "kmod-gpio-button-hotplug", "kmod-ipt-offload", "adblock", "auc", "fwtool", "hostapd-common", "jshn", "jsonfilter", "kmod-ip6tables", "kmod-ipt-conntrack", "kmod-ipt-core", "kmod-ipt-nat", "kmod-lib-crc-ccitt", "kmod-nf-conntrack", "kmod-nf-conntrack6", "kmod-nf-ipt", "kmod-nf-ipt6", "kmod-nf-nat", "kmod-nf-reject", "kmod-nf-reject6", "kmod-ppp", "kmod-pppoe", "kmod-pppox", "kmod-slhc", "libblobmsg-json", "libip4tc", "libip6tc", "libiwinfo", "libiwinfo-lua", "libjson-c", "libjson-script", "liblua", "liblucihttp", "liblucihttp-lua", "libnl-tiny", "libopenssl", "libubox", "libubus", "libubus-lua", "libuci", "libuclient", "libxtables", "lua", "luci", "luci-app-adblock", "luci-app-attendedsysupgrade", "luci-app-firewall", "luci-app-sqm", "luci-app-wifischedule", "luci-base", "luci-lib-ip", "luci-lib-jsonc", "luci-lib-nixio", "luci-mod-admin-full", "luci-proto-ipv6", "luci-proto-ppp", "luci-theme-bootstrap", "nano", "openwrt-keyring", "procd", "rpcd", "rpcd-mod-rrdns", "sqm-scripts", "tc", "uboot-envtools", "ubox", "ubus", "ubusd", "uhttpd", "usign", "wget", "kmod-nf-flow", "ubi-utils", "kernel", "kmod-mwlwifi", "mwlwifi-firmware-88w8864"], "upgrades": {"luci-app-wifischedule": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-lib-ip": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-mod-admin-full": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-proto-ppp": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-app-adblock": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-theme-bootstrap": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-lib-nixio": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-app-attendedsysupgrade": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-base": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-app-firewall": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "dnsmasq": ["2.80-1.4", "2.80test3-1"], "luci-lib-jsonc": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"], "luci-proto-ipv6": ["git-19.181.61051-9bab4ba-1", "git-19.188.21194-7d8df17-1"]}, "version": "18.06.4"}

aparcar, 2019-08-14:
I think you just send the response from the server and not the request your device does.

apcimino, 2019-08-14:
Sorry about that. Below are the json requests my browser sends. Let me know if you need something else.

{"jsonrpc":"2.0","id":0,"result":[0,{"packages":{"iwinfo":"2018-07-24-94b1366d-1","luci-app-adblock":"git-19.188.21194-7d8df17-1","opkg":"2019-01-18-7708a01a-1","wget":"1.19.5-1","ubus":"2018-10-06-221ce7e7-1","rpcd":"2018-11-28-3aa81d0d-2","busybox":"1.28.4-3","luci-lib-ip":"git-19.188.21194-7d8df17-1","libubus-lua":"2018-10-06-221ce7e7-1","kmod-nf-reject6":"4.14.63-1","nano":"4.3-1","libiwinfo-lua":"2018-07-24-94b1366d-1","swconfig":"11","libiwinfo":"2018-07-24-94b1366d-1","kmod-nf-flow":"4.14.63-1","kmod-lib-crc-ccitt":"4.14.63-1","luci-theme-bootstrap":"git-19.188.21194-7d8df17-1","kmod-pppoe":"4.14.63-1","kmod-pppox":"4.14.63-1","kmod-ipt-conntrack":"4.14.63-1","kmod-nf-reject":"4.14.63-1","base-files":"192-r7258-5eb055306f","kmod-nf-nat":"4.14.63-1","netifd":"2019-01-31-a2aba5c7-2.1","uboot-envtools":"2018.03-1","adblock":"3.5.5-3","dnsmasq":"2.80test3-1","procd":"2018-03-28-dfb68f85-1","ubusd":"2018-10-06-221ce7e7-1","firewall":"2018-08-13-1c4d5bcd-1","luci-app-firewall":"git-19.188.21194-7d8df17-1","kmod-nf-ipt":"4.14.63-1","mwlwifi-firmware-88w8864":"2018-11-14-81413aa9-1","tc":"4.16.0-8","libuci":"2019-05-17-f199b961-1","liblua":"5.1.5-1","libip4tc":"1.6.2-1","ubi-utils":"2.0.2-1","kmod-ip6tables":"4.14.63-1","odhcp6c":"2018-07-14-67ae6a71-15","fstools":"2018-04-16-e2436836-1","uclient-fetch":"2018-11-24-3ba74ebc-1","libuclient":"2018-11-24-3ba74ebc-1","uci":"2019-05-17-f199b961-1","lua":"5.1.5-1","wpad-mini":"2018-05-21-62566bc2-6","dropbear":"2017.75-7.1","mtd":"23","odhcpd-ipv6only":"1.15-3","libjson-c":"0.12.1-2","luci-app-sqm":"1.2.4-1","libip6tc":"1.6.2-1","luci-proto-ppp":"git-19.188.21194-7d8df17-1","ppp":"2.4.7-12","luci-mod-admin-full":"git-19.188.21194-7d8df17-1","libubox":"2018-07-25-c83a84af-2","luci-base":"git-19.188.21194-7d8df17-1","kmod-gpio-button-hotplug":"4.14.63-2","logd":"2018-02-14-128bc35f-2","libjson-script":"2018-07-25-c83a84af-2","luci-proto-ipv6":"git-19.188.21194-7d8df17-1","libblobmsg-json":"2018-07-25-c83a84af-2","iptables":"1.6.2-1","openwrt-keyring":"2018-05-18-103a32e9-1","luci-app-attendedsysupgrade":"git-19.188.21194-7d8df17-1","sqm-scripts":"1.2.4-1","jshn":"2018-07-25-c83a84af-2","kmod-mwlwifi":"4.14.63+2018-06-15-8683de8e-1","kmod-ipt-core":"4.14.63-1","liblucihttp":"2019-06-05-91c01c3c-1","luci-app-wifischedule":"git-19.188.21194-7d8df17-1","kmod-ppp":"4.14.63-1","libubus":"2018-10-06-221ce7e7-1","uhttpd":"2018-11-28-cdfc902a-3","auc":"0.0.9-2","kmod-nf-conntrack":"4.14.63-1","usign":"2015-07-04-ef641914-1","ip6tables":"1.6.2-1","libxtables":"1.6.2-1","kmod-nf-ipt6":"4.14.63-1","luci-lib-nixio":"git-19.188.21194-7d8df17-1","liblucihttp-lua":"2019-06-05-91c01c3c-1","luci-lib-jsonc":"git-19.188.21194-7d8df17-1","luci":"git-19.188.21194-7d8df17-1","kmod-nf-conntrack6":"4.14.63-1","ubox":"2018-02-14-128bc35f-2","kernel":"4.14.63-1-c0354e0486370ffeacebcc4352595371","libnl-tiny":"0.1-5","fwtool":"1","jsonfilter":"2018-02-04-c7e938d6-1","hostapd-common":"2018-05-21-62566bc2-6","kmod-ipt-offload":"4.14.63-1","libopenssl":"1.0.2s-1","kmod-slhc":"4.14.63-1","rpcd-mod-rrdns":"20170710","ppp-mod-pppoe":"2.4.7-12","kmod-ipt-nat":"4.14.63-1"}}]}
{"jsonrpc":"2.0","id":1,"result":[0,{"kernel":"4.14.63","hostname":"OpenWrt","system":"ARMv7 Processor rev 1 (v7l)","model":"Linksys WRT1900ACS","board_name":"armada-385-linksys-shelby","release":{"distribution":"OpenWrt","version":"18.06.1","revision":"r7258-5eb055306f","target":"mvebu\/cortexa9","description":"OpenWrt 18.06.1 r7258-5eb055306f"}}]}
{"jsonrpc":"2.0","id":4,"result":[0,{"localtime":1565794387,"uptime":88232,"load":[2528,1824,32],"memory":{"total":524214272,"free":462307328,"shared":1736704,"buffered":1806336},"swap":{"total":0,"free":0}}]}
{"jsonrpc":"2.0","id":5,"result":[0,{"value":"https:\/\/chef.libremesh.org\/"}]}
{"jsonrpc":"2.0","id":6,"result":[0,{"value":"1"}]}

aparcar, 2020-02-25:
The upgrade requests have changed and luci-app-attendedsysupgrade needs to be rewritten. I'll do that soonish, which makes this issue obsolete.

closed 2020-02-25 by aparcar