openwrt/asu #179: Allow pkg shadowsocks-libev to be added to image

mrkaffeine92, 2019-07-11:
Have been unable to compile anything on the SDK all day today, and all I need is shadowsocks-libev on a particular image.

Please add it to the image creator.

Thanks

aparcar, 2019-07-11:
@mrkaffeine92 is shadowsocks-libev available as a regular OpenWrt image in packages.git?

mrkaffeine92, 2019-07-11:
No, not yet.

 If that is all that you will be allowing inside the images then I apologize and thanks for your response

aparcar, 2019-07-11:
@mrkaffeine92 The question is why isn't your packages available via the official repositories, I'll look into it.

aparcar, 2019-07-11:
The package seem to be available:

```
shadowsocks-client - 0.6-b59d754f838213d60b908aed0b7d4d5a81f273e2 - shadowsocks client for router
shadowsocks-libev-config - 3.1.3-3 - shadowsocks-libev config scripts
shadowsocks-libev-ss-local - 3.1.3-3 - shadowsocks-libev ss-local
shadowsocks-libev-ss-redir - 3.1.3-3 - shadowsocks-libev ss-redir
shadowsocks-libev-ss-rules - 3.1.3-3 - shadowsocks-libev ss-rules
shadowsocks-libev-ss-server - 3.1.3-3 - shadowsocks-libev ss-server
shadowsocks-libev-ss-tunnel - 3.1.3-3 - shadowsocks-libev ss-tunnel
```

What distro/release are you testing?

mrkaffeine92, 2019-07-12:
These are the instructions I have been following, I continuously get errors. Have already patched kernel, following different guides, tried building via with image-builder.

[code]
#Download either the full SDK OpenWRT image, or generic for your card type found by Googling your Router+'chipset'
#Unpack the file in to a Home directory
Tar xjf OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2
#Enter Directory
Cd OpenWrt-SDK-ar71xx-*
# Add feeds
./scripts/feeds update -a
git clone https://github.com/shadowsocks/openwrt-feeds.git
#Create a new directory in: /OpenWRT_SDK*/package/network/ called shadowsocks
#Copy the two folders from the shadowsocks feeds to this folder (base, package)
#In folder base, add 'lib' to the beginning of mbedtls to make it "libmbedtls"
#In folder packages, add 'lib' to the beginning of pcre to create "libpcre" 
# Get shadowsocks-libev Makefile
git clone https://github.com/shadowsocks/openwrt-shadowsocks.git 
#Copy the file Makefile to the main directory of shadowsocks you created previously
Make menuconfig
#Select the package you wish to compile, under Network - shadowsocks and shadowsocks Server
#Compile the package
make package/shadowsocks-libev/compile V=99[/code]

I have tried on Ubuntu, Kali, and Debian. Used the master as well as generic, 18.0.6.4 and 18.0.6

If there is a way to get this compiled, please keep me posted.

mrkaffeine92, 2019-07-12:
I figured out why my packages weren't compiling. I had to manually download the generic for my build in to my SDK folder. Anyway, it would be nice if your image compiler would add packages via specify the github link and build its own dependencies etc or the user could add them.

closed 2019-07-12 by mrkaffeine92

aparcar, 2019-07-12:
Adding external repos adds to much complexity, at least if it's done on demand. Please add your package to the official packages repository