openwrt/asu #180: Ubiquiti EdgeRouter X - incorrect profile

SirToffski, 2019-07-14:
[Chef ImageBuilder](https://chef.libremesh.org/) is using an outaded profile for Ubiquiti EdgeRouter X on Snapshots.

### Way to replicate the issue

#### In [Chef ImageBuilder](https://chef.libremesh.org/)

- In [Chef ImageBuilder](https://chef.libremesh.org/), select
- Distribution: openwrt
- Version: snapshot
- Device model:  EdgeRouter - [Ubiquiti EdgeRouter X (ramips/mt7621)] 

Build will show an error `Incompatible package selection.` Build-log file attached
[failed-build-log.txt](https://github.com/aparcar/attendedsysupgrade-server/files/3390252/failed-build-log.txt)

Build-log shows the following info:
```
TARGET="ramips/mt7621"
PROFILE="ubnt-erx"
VERSION="snapshot"
PACKAGES_HASH="cb076f81fa82"
PACKAGES="*** [edited for clarity]"
IMAGE_HASH="None"
DISTRO="openwrt"
REQUEST_HASH="d699ae1b0dc4"
DEFAULTS_HASH=""
REQUEST_ID="7340"
./meta image
```

#### Cloning the meta-imagebuilder repository

When cloning the repository and attempting the same build, the following output is seen. Do note the `Profile "ubnt-erx" does not exist!` error.

```
 $ export TARGET="ramips/mt7621"
 $ export PROFILE="ubnt-erx"
 $ export VERSION="snapshot"
 $ export PACKAGES_HASH="1d9b8b8c71e9"
 $ export PACKAGES="*** [edited for clarity]"
 $ export IMAGE_HASH="None"
 $ export DISTRO="openwrt"
 $ export REQUEST_HASH="94fd39e07135"
 $ ./meta image

-----//------

openwrt-imagebuilder- 100%[======================>]  55.46M  11.7MB/s    in 5.4s    

2019-07-14 14:01:26 (10.2 MB/s) - ‘openwrt-imagebuilder-ramips-mt7621.Linux-x86_64.tar.xz’ saved [58150820/58150820]

Checking sha256sum a second time
openwrt-imagebuilder-ramips-mt7621.Linux-x86_64.tar.xz: OK
ImageBuilder is up-to-date
Profile "ubnt-erx" does not exist!
Use "make info" to get a list of available profile names.
make[1]: *** [Makefile:190: _check_profile] Error 1
make: *** [Makefile:195: image] Error 2
```

### Resolving the issue

In the [latest snapshot download](https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/) directory, Ubiquiti images are named:
```
ubiquiti_edgerouterx-initramfs-kernel.bin 
ubiquiti_edgerouterx-sfp-initramfs-kernel.bin 
ubiquiti_edgerouterx-sfp-squashfs-sysupgrade.bin 
ubiquiti_edgerouterx-squashfs-sysupgrade.bin
```

Repeating build process with `export PROFILE="ubiquiti_edgerouterx"` has resulted in successful image generation.

```
 $ export TARGET="ramips/mt7621"
 $ export PROFILE="ubiquiti_edgerouterx"
 $ export VERSION="snapshot"
 $ export PACKAGES_HASH="1d9b8b8c71e9"
 $ export PACKAGES="*** [edited for clarity]"
 $ export IMAGE_HASH="None"
 $ export DISTRO="openwrt"
 $ export REQUEST_HASH="94fd39e07135"
 $ ./meta image

 -----//------

Building images...
Parallel mksquashfs: Using 1 processor
Creating 4.0 filesystem on /home/toffski/bifrost/chef/meta-imagebuilder/imagebuilder/openwrt/snapshot/ramips/mt7621/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/home/toffski/bifrost/chef/meta-imagebuilder/imagebuilder/openwrt/snapshot/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/dev".
Ignoring, exclude it (-e/-ef) to override.
[==================================================================|] 1580/1580 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 8505.69 Kbytes (8.31 Mbytes)
	39.67% of uncompressed filesystem size (21443.60 Kbytes)
Inode table size 16428 bytes (16.04 Kbytes)
	22.17% of uncompressed inode table size (74104 bytes)
Directory table size 21864 bytes (21.35 Kbytes)
	45.08% of uncompressed directory table size (48504 bytes)
Number of duplicate files found 169
Number of inodes 2183
Number of files 1545
Number of fragments 39
Number of symbolic links  482
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 155
Number of ids (unique uids + gids) 1
Number of uids 1
	unknown (0)
Number of gids 1
	unknown (0)
sysupgrade-ubiquiti_edgerouterx/
sysupgrade-ubiquiti_edgerouterx/CONTROL
sysupgrade-ubiquiti_edgerouterx/kernel
sysupgrade-ubiquiti_edgerouterx/root

Calculating checksums...
```

Camis, 2019-08-05:
Fixed

aparcar, 2019-08-05:
So the issue is resolved or still a problem for Chef?

Camis, 2019-08-05:
Still:

https://chef.libremesh.org/download/faillogs/faillog-ab447e36732f.txt

aparcar, 2019-08-14:
Please try again via https://aparcar.stephen304.com/

SirToffski, 2019-08-16:
Image built successfully using https://aparcar.stephen304.com/

Thanks!

closed 2019-08-16 by aparcar