openwrt/asu #181: Problem with creating an image in line with a snapshot version

podtor, 2019-07-31:
I tried creating an image in line with a snapshot version for tp-link 842nd v1 and r6200 and d-link 118 A1 but always received warning "Incompatible package selection. See build log for details Build log"

```
example 1
* opkg_install_pkg: Package size mismatch: wpad-basic is 288613 bytes, expecting 288619 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
```

```
example 2
/home/aparcar/server/worker/imagebuilder/openwrt/snapshot/ramips/mt7620/dl/imagebuilder
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt76x0e:
 * 	wireless-regdb
 * opkg_install_cmd: Cannot install package kmod-mt76x0e.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
```

Camis, 2019-08-02:
I have this problem too:

https://chef.libremesh.org/download/faillogs/faillog-833f8680eb49.txt

aparcar, 2019-08-02:
I think there were some issues upstream. Please retry in a day and close this issue if solved.

podtor, 2019-08-02:
Still the same problem with snapshot

d-link 118 A1
https://chef.libremesh.org/download/faillogs/faillog-b779c141f453.txt

tp-link wr842nd v1 (ar71xx)
https://chef.libremesh.org/download/faillogs/faillog-0770250f22ce.txt

aparcar, 2019-08-17:
This should be (finally) fixed via https://github.com/aparcar/openwrt-metabuilder/commit/62661b5336e3a25e7b252c7bcca6ed1f273169f0 and is deployed on https://aparcar.stephen304.com/, please give it a try!

amq, 2019-09-28:
This problem is currently present on https://aparcar.stephen304.com/, see https://aparcar.stephen304.com//download/faillogs/faillog-e8a264245f27.txt

aparcar, 2019-09-28:
The ar71xx targets are all very outdated and will no longer be build. Could you please retry with the succeeding target ath79?

amq, 2019-09-28:
Does this only affect the online version or also the self-hosted one? I'll soon have to maintain a whole bunch of MikroTik, TP-Link, Ubiquiti and GL.iNet routers, all with ar71xx.

aparcar, 2020-02-25:
This is currently an upstream issue, see the current [buildlog](https://chef.libremesh.org/store/SNAPSHOT/ath79/generic/tplink_tl-wr842n-v1/e3b0c44298fc/buildlog.txt)

aparcar, 2020-03-05:
> Does this only affect the online version or also the self-hosted one? I'll soon have to maintain a whole bunch of MikroTik, TP-Link, Ubiquiti and GL.iNet routers, all with ar71xx.

Can you consider upgrading to ath79? The custom build does not support ar71xx as it is disabled in SNAPSHOTS and that's the only supported version right now.

aparcar, 2020-03-05:
Builds work again at chef.libremesh.org

[image]

closed 2020-03-05 by aparcar