openwrt/asu #182: Attended Sysupgrade update to old version of package

podtor, 2019-07-31:
Why "attended sysupgrade" wanted to update to an old version of a package?

```
Package upgrades available
luci-ssl-openssl: git-19.189.59078-9c5162d-1 to git-19.181.61051-9bab4ba-1
luci-app-attendedsysupgrade: git-19.190.55614-35357e4-1 to git-19.181.61051-9bab4ba-1
luci-proto-relay: git-19.189.59078-9c5162d-1 to git-19.181.61051-9bab4ba-1
```

aparcar, 2019-08-08:
I'm guessing the updater was outdated. Do you have the request id?

bigwave, 2019-08-12:
I too seem to be getting this error.

If the "request id" that you are looking for is the one that is returned when the "Request Firmware" button is pressed in Luci, then, for me, it is `{"request_hash": "370c31fad131"}`

[image]

aparcar, 2019-08-12:
Thanks, I'm currently rewriting large parts of the imageserver to handle such problems better, please give me some time.

aparcar, 2020-03-07:
Please check the latest luci app it should work again. If not reopen this issue please.

closed 2020-03-07 by aparcar