openwrt/asu #188: ar71xx/generic missing

jchulce, 2019-08-16:
It appears that ar71xx/generic has disappeared from https://aparcar.stephen304.com/ . Devices like the ubiquiti unifi uap-pro and others I've spot checked are missing ar71xx versions. The target is still on https://chef.libremesh.org/ . I don't see any spelling errors in /asu/utils/distributions/openwrt/distro_config.yml like there was for #165

jow-, 2019-08-16:
The link you mentioned states "Warning: only snapshot builds here". OpenWrt master dropped ar71xx in favor to the new ath79 target.

jchulce, 2019-08-16:
Thanks jow. Looks like many ar71xx openwrt targets have not been ported to ath79, thus are not in snapshots. I was unaware that all those devices were dropped. This includes uap-pro and the others I checked. Not an asu issue

closed 2019-08-16 by jchulce