openwrt/asu #195: Enhancement: toggle to switch between tar/bin sysupgrade file

ich0x, 2019-09-05:
Snapshots generated at aparcar.stephen304.com for devices no longer generate *-sysupgrade.tar images which can be used in Luci which complains of the invalid format of the *.bin currently generated.

 I suggest implementing a toggle to switch between the two formats.

If you need logs compare
https://aparcar.stephen304.com/download/openwrt/snapshot/ramips/mt7621/netgear_r6220/c1fad73df0cfc65/openwrt-c1fad73df0cfc65-ramips-mt7621-netgear_r6220.log

to

https://chef.libremesh.org/download/openwrt/18.06.4/ramips/mt7621/r6220/d6d111f92b6a165/buildlog-c6d2149a5757d5a.txt

Camis, 2019-09-06:
Luci accepts .bin files without any problems.

I have 6220 too.

ich0x, 2019-09-06:
[image]
Can you try this one? Just to confirm something is broken or not?
https://aparcar.stephen304.com/download/openwrt/snapshot/ramips/mt7621/netgear_r6220/eeb44eebeb221db/

Camis, 2019-09-06:
> Can you try this one?

Working ok:

[image]

ich0x, 2019-09-06:
I guessing it is a Luci 18.06 thing then. That bin will not load for me.

ich0x, 2019-09-08:
None of the snapshots lately are working on my stable 18.06.4 
I did the following while ssh, I couldn't wget from the https stephen304 as I don't have SSL libs so I hosted locally >>
root@OpenWrt:/tmp# DOWNLOAD_LINK="http://192.168.1.250:8081/9988/openwrt/openwrt-f8cd479543d714c-ramip
s-mt7621-netgear_r6220-squashfs-sysupgrade.bin"; SHA256SUMS="http://192.168.1.250:8081/9988/openwrt/sh
a256sums"
root@OpenWrt:/tmp# cd /tmp;wget $DOWNLOAD_LINK;wget $SHA256SUMS;sha256sum -c sha256sums 2>/dev/null|gr
ep OK
Downloading 'http://192.168.1.250:8081/9988/openwrt/openwrt-f8cd479543d714c-ramips-mt7621-netgear_r622
0-squashfs-sysupgrade.bin'
Connecting to 192.168.1.250:8081
Writing to 'openwrt-f8cd479543d714c-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.bin'
openwrt-f8cd479543d7 100% |*******************************|  5100k  0:00:00 ETA
Download completed (5222645 bytes)
Downloading 'http://192.168.1.250:8081/9988/openwrt/sha256sums'
Connecting to 192.168.1.250:8081
Writing to 'sha256sums'
sha256sums           100% |*******************************|   807   0:00:00 ETA
Download completed (807 bytes)
openwrt-f8cd479543d714c-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.bin: OK
root@OpenWrt:/tmp# sysupgrade -v /tmp/*.bin
Invalid sysupgrade file.
Image check 'platform_check_image' failed.

The built images are from https://aparcar.stephen304.com/download/openwrt/snapshot/ramips/mt7621/netgear_r6220/f8cd479543d714c

closed 2019-09-08 by ich0x