openwrt/asu #196: Netgear R6220 - Snapshot "Unsupported device, release, target, subtraget or board"

Camis, 2019-09-05:
`distro: "OpenWrt"
installed: {luci-proto-wireguard: "git-19.246.62933-f9074ad-1", luci-theme-material: "git-19.246.62933-f9074ad-1",…}
base-files: "205-r10956-62dbe361a1"
busybox: "1.31.0-1"
dnsmasq: "2.80-14"
dropbear: "2019.78-2"
firewall: "2019-08-21-bf29c1e7-1"
fstools: "2019-08-28-6a61b9ac-1"
ip6tables: "1.8.3-2"
iptables: "1.8.3-2"
iwinfo: "2019-09-01-f599a8dc-1"
kernel: "4.14.141-1-327be771dfe608186b40d57505fffded"
kmod-gpio-button-hotplug: "4.14.141-3"
kmod-ipt-offload: "4.14.141-1"
kmod-leds-gpio: "4.14.141-1"
kmod-mt76x2: "4.14.141+2019-08-22-2a0edbb4-1"
kmod-mt7603: "4.14.141+2019-08-22-2a0edbb4-1"
kmod-usb3: "4.14.141-1"
kmod-usb-ledtrig-usbport: "4.14.141-1"
libc: "1.1.23-2"
logd: "2019-06-16-4df34a4d-3"
luci-app-attendedsysupgrade: "git-19.246.62933-f9074ad-1"
luci-app-wireguard: "git-19.246.62933-f9074ad-1"
luci-proto-wireguard: "git-19.246.62933-f9074ad-1"
luci-ssl: "git-19.246.62933-f9074ad-1"
luci-theme-material: "git-19.246.62933-f9074ad-1"
luci-theme-rosy: "git-19.246.62933-f9074ad-1"
mtd: "25"
netifd: "2019-08-05-5e02f944-1"
odhcp6c: "2019-01-11-d2e247d8-16"
odhcpd-ipv6only: "2019-05-17-41a74cba-3"
opkg: "2019-06-14-dcbc142e-1"
ppp: "2.4.7.git-2019-05-25-2"
ppp-mod-pppoe: "2.4.7.git-2019-05-25-2"
swconfig: "12"
uci: "2019-09-01-415f9e48-3"
uclient-fetch: "2019-05-30-3b3e368d-1"
urandom-seed: "1.0-1"
urngd: "2019-06-17-c057e177-1"
wireguard: "0.0.20190702-1"
wireguard-tools: "0.0.20190702-1"
wpad-wolfssl: "2018-12-02-c2c6c01b-7"
revision: "r10956-62dbe361a1"
target: "ramips/mt7621"
upgrade_packages: "1"
version: "SNAPSHOT"`

aparcar, 2020-02-25:
This is currently an upstream issue see [buildlog](https://chef.libremesh.org/store/SNAPSHOT/ramips/mt7621/netgear_r6220/e3b0c44298fc/buildlog.txt)

aparcar, 2020-03-05:
Works again at https://chef.libremesh.org/

[image]

closed 2020-03-05 by aparcar