openwrt/asu #197: 19.07 snapshot error after selecting device

patsy1912, 2019-09-06:
**Describe the bug**
A clear and concise description of what the bug is.

**Server URL**
Please write the URL of the server you use.

**JSON request**
Please post the JSON you've send to the server.

**Build Log** 
The server should return a build log, please post a link to the build log below.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Reproduce locally**
In case you could retrieve the build log, please run the lines under *BUILD COMMAND* via the [openwrt-metabuilder script](https://github.com/aparcar/meta-imagebuilder). Also possible via **Docker**

aparcar, 2019-09-08:
Please provide some additional information as suggested by the issue template

patsy1912, 2019-09-08:
[image]
[image]
[image]

Sorry for not being clearer

On the new frontend when selecting 1907 snapshot and then selecting which device to compile for unexpected error pops up,trired on multi browsers and multiple platforms and received same error message

closed 2019-09-11 by patsy1912