openwrt/asu #198: Linksys EA8300 snapshots (New FrontEnd)

syto203, 2019-09-14:
**Describe the bug**
unable to choose linksys ea8300 in the new frontend [https://sudhanshu16.github.io/openwrt-firmware-selector)/](https://sudhanshu16.github.io/openwrt-firmware-selector)/) works normally on [https://aparcar.stephen304.com](https://aparcar.stephen304.com) .

the new frontend page throws the error "An unexpected error occurred. Please try again"
and the console reports "Request failed with status code 404".

ich0x, 2019-09-14:
@syto203 The server was down when you wrote the issue, and it remains down as of now.

aparcar, 2019-09-15:
I'll have to rebuild all the images, will do that later

aparcar, 2020-02-24:
Works on the new server https://chef.libremesh.org/
[image]

closed 2020-02-24 by aparcar