openwrt/asu #200: [luci-app-attendedsysupgrade] Nothing displayed on the web UI

claell, 2019-09-17:
**Describe the bug**
I installed `luci-app-attendedsysupgrade` to my router (Xiaomi Router 3G) running the latest stable OpenWrt version using opkg. A menu entry got added, but unfortunately there is nothing displayed when clicking on it.

**Expected behavior**
The web UI is displayed as shown on the screenshot in the README.md

**Screenshots**
Currently I don't have the package installed anymore, but I can create one if needed.

Camis, 2019-09-26:
Are you using a custom LUCI theme ?.

claell, 2019-09-28:
I do (`luci-theme-material`). I thought I was also experiencing it with the default one, but it turns out that it works there. Thanks for that hint.

So this is apparently related to the theme. Do you think it is a bug in the theme or in this application?

Screenshots:
[image]

[image]

aparcar, 2020-03-07:
The app was recently updated to the new API of the server and should work again. If not please reoopen the issue.

claell, 2020-03-09:
Alright, thanks! I will try to give feedback when I tested.

tlaurion, 2020-03-09:
Testing

tlaurion, 2020-03-09:
I guess server should be https://images.aparcar.org ?

aparcar, 2020-03-09:
No I disabled images.aparcar.org as the server at chef.libremesh.org is much faster.

tlaurion, 2020-03-09:
@aparcar : "Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://images.aparcar.org/api/upgrade-check. (Reason: CORS request did not succeed)."

attendedsysupgrade-common 0.1-2
luci-app-attendedsysupgrade git-20.064.26341-4fc4c79-1

Is that right?

tlaurion, 2020-03-09:
> chef.libremesh.org

Upgrade server down or could not connect

tlaurion, 2020-03-09:
> No I disabled images.aparcar.org as the server at chef.libremesh.org is much faster.

Please document somewhere for users to be able to know where to connect.... 6 times I open issues on that already.

How can  I help troubleshoot.
Another repo needs to be specified to use current luci-app-attendedsysupgrade and its corresponding attendedsysupgrade-common dep?

aparcar, 2020-03-09:
>  Gives 404

Yes, sorry the API changed and only the `luci-app-attendedsysupgrade` from snapshots works for now. This is due to missing JSON files in previous builds. I'm trying to get those changes upstream, however it take some time...

Meaning, I could backport the current LuCI app to 19.07/18.06, however you could only upgrade to unstable snapshots.

> Please document somewhere for users to be able to know where to connect.... 6 times I open issues on that already.

Yes I know this is tedious, the [UPC](https://upc.edu) provides a fast server for ASU, however I lost access for a while. In the meantime I setup a temporary VM for testing, sorry for the confusion. The domain `chef.libremesh.org` should be stable now.

tlaurion, 2020-03-09:
@aparcar : anything that works for the moment :)
Just give instructions and I will follow through.

aparcar, 2020-03-09:
@tlaurion I don't want to motivate users to run snapshot builds. I'll try to get the functionality in the next stable release!

tlaurion, 2020-03-10:
@aparcar: sonifni read between the lines, if I reflash a snapshot, luci and common should be able to pick up and I should be able to test. Right?

aparcar, 2020-03-10:
Yes, for testing snapshots are great, but it happens like once a week that a random device breaks. So it should be your mains connection.

tlaurion, 2020-05-26:
@aparcar Tell me if you want me to open up a new ticket.

Generated a new snapshot image that i've flashed from http://chef.libremesh.org/ :
- Linksys WRT3200ACM / Linksys Rango
- Packages:
```
attendedsysupgrade-common
base-files
busybox
cgi-io
collectd
collectd-mod-cpu
collectd-mod-interface
collectd-mod-iwinfo
collectd-mod-load
collectd-mod-memory
collectd-mod-network
collectd-mod-rrdtool
dnsmasq
dropbear
firewall
fstools
fwtool
getrandom
hostapd-common
ip6tables
iptables
iwinfo
jshn
jsonfilter
kernel
kmod-bluetooth
kmod-btmrvl
kmod-cfg80211
kmod-crypto-aead
kmod-crypto-cmac
kmod-crypto-ecb
kmod-crypto-ecdh
kmod-crypto-hash
kmod-crypto-kpp
kmod-crypto-manager
kmod-crypto-null
kmod-crypto-pcompress
kmod-gpio-button-hotplug
kmod-gre
kmod-hid
kmod-input-core
kmod-input-evdev
kmod-ip6tables
kmod-ipt-conntrack
kmod-ipt-core
kmod-ipt-nat
kmod-ipt-offload
kmod-ipt-raw
kmod-iptunnel
kmod-lib-crc-ccitt
kmod-lib-crc16
kmod-lib-textsearch
kmod-mac80211
kmod-mmc
kmod-mwifiex-sdio
kmod-mwlwifi
kmod-nf-conntrack
kmod-nf-conntrack-netlink
kmod-nf-conntrack6
kmod-nf-flow
kmod-nf-ipt
kmod-nf-ipt6
kmod-nf-nat
kmod-nf-nathelper-extra
kmod-nf-reject
kmod-nf-reject6
kmod-nfnetlink
kmod-nls-base
kmod-ppp
kmod-pppoe
kmod-pppox
kmod-pptp
kmod-regmap-core
kmod-slhc
kmod-usb-core
libblobmsg-json
libc
libgcc1
libip4tc2
libip6tc2
libiwinfo-lua
libjson-script
libltdl7
liblua5.1.5
liblucihttp-lua
liblucihttp0
libnl-tiny
libpthread
librrd1
libubus-lua
logd
lua
luci
luci-app-attendedsysupgrade
luci-app-firewall
luci-app-nlbwmon
luci-app-opkg
luci-app-statistics
luci-base
luci-compat
luci-lib-ip
luci-lib-iptparser
luci-lib-jsonc
luci-lib-nixio
luci-mod-admin-full
luci-mod-network
luci-mod-status
luci-mod-system
luci-proto-ipv6
luci-proto-ppp
luci-theme-bootstrap
mtd
mwifiex-sdio-firmware
netifd
nlbwmon
odhcp6c
odhcpd-ipv6only
openwrt-keyring
opkg
ppp
ppp-mod-pppoe
procd
rpcd
rpcd-mod-file
rpcd-mod-iwinfo
rpcd-mod-luci
rpcd-mod-rpcsys
rpcd-mod-rrdns
rrdtool1
swconfig
ubi-utils
uboot-envtools
ubox
ubus
ubusd
uci
uclient-fetch
uhttpd
uhttpd-mod-ubus
urandom-seed
urngd
usign
wireless-regdb
wpad-basic
zlib
```

Flash successful.

Then from Attended Sysupgrade:
- Server: `https://chef.libremesh.org`
- Search for upgrades:
```
New release snapshot available
Installed version: SNAPSHOT
```
- Request image:
`Unsupported profile: linksys_rango`

Details:
F12 in browser:
- Request:
`{"version":"snapshot","diff_packages":true,"profile":"linksys,rango","packages":["iwinfo","luci-app-statistics","liblucihttp0","kmod-lib-textsearch","cgi-io","libc","luci-app-nlbwmon","opkg","kmod-usb-core","luci-app-opkg","wpad-basic","ubus","libpthread","kmod-crypto-manager","kmod-crypto-pcompress","rpcd","busybox","kmod-btmrvl","luci-lib-ip","kmod-nfnetlink","libubus-lua","kmod-crypto-hash","kmod-nf-reject6","kmod-crypto-aead","libiwinfo-lua","swconfig","luci-mod-system","kmod-nf-flow","kmod-lib-crc-ccitt","getrandom","kmod-gre","luci-theme-bootstrap","kmod-pppoe","kmod-pppox","kmod-ipt-conntrack","kmod-nf-reject","base-files","kmod-lib-crc16","kmod-nf-nat","kmod-input-core","netifd","kmod-ipt-raw","uboot-envtools","dnsmasq","procd","nlbwmon","mwifiex-sdio-firmware","ubusd","kmod-iptunnel","luci-mod-status","kmod-pptp","collectd-mod-interface","firewall","luci-lib-iptparser","luci-app-firewall","kmod-nf-ipt","collectd-mod-network","kmod-mmc","ubi-utils","kmod-ip6tables","odhcp6c","fstools","uclient-fetch","uci","lua","collectd","libip4tc2","dropbear","kmod-input-evdev","kmod-crypto-kpp","rrdtool1","kmod-crypto-ecdh","luci-compat","rpcd-mod-file","mtd","odhcpd-ipv6only","kmod-mwifiex-sdio","rpcd-mod-luci","urandom-seed","luci-proto-ppp","luci-mod-admin-full","ppp","luci-base","kmod-gpio-button-hotplug","logd","kmod-bluetooth","kmod-mac80211","collectd-mod-iwinfo","libjson-script","luci-proto-ipv6","libblobmsg-json","libip6tc2","kmod-crypto-ecb","iptables","openwrt-keyring","luci-app-attendedsysupgrade","kmod-nf-conntrack-netlink","kmod-nf-nathelper-extra","jshn","kmod-mwlwifi","kmod-ipt-core","attendedsysupgrade-common","kmod-ppp","kmod-crypto-null","collectd-mod-load","uhttpd","kmod-nf-conntrack","usign","ip6tables","liblua5.1.5","zlib","kmod-nf-ipt6","luci-lib-nixio","rpcd-mod-rpcsys","collectd-mod-rrdtool","liblucihttp-lua","kmod-crypto-cmac","luci-lib-jsonc","collectd-mod-memory","luci","kmod-nf-conntrack6","libgcc1","librrd1","ubox","kernel","libnl-tiny","rpcd-mod-iwinfo","luci-mod-network","libltdl7","kmod-regmap-core","kmod-nls-base","uhttpd-mod-ubus","fwtool","jsonfilter","hostapd-common","collectd-mod-cpu","kmod-ipt-offload","kmod-hid","wireless-regdb","urngd","kmod-slhc","kmod-cfg80211","rpcd-mod-rrdns","ppp-mod-pppoe","kmod-ipt-nat"]}`

Response:
`{"message":"Unsupported profile: linksys_rango","status":"bad_profile"}
`

aparcar, 2020-05-26:
What a perfect bug report, thank you! I know how to fix the problem and let you know once I have a fix!

aparcar, 2020-05-27:
Implemented via #250 

Tested with the same device and it does seem to work

[image]

closed 2020-05-28 by aparcar

tlaurion, 2020-08-09:
@aparcar [Redoing the above gives](https://github.com/aparcar/asu/issues/200#issuecomment-634084068):

```
Build
Model: Linksys WRT3200ACM / Linksys Rango
Platform: mvebu/cortexa9
Version: SNAPSHOT (r13673-f786de2095)
Date: Sat, 04 Jul 2020 22:00:59 GMT

```

Also results in a 404 error when attempting to reach https://chef.libremesh.org/api/build

tlaurion, 2020-08-09:
@aparcar Please reopen issue.