openwrt/asu #201: Upgrade server down or could not connect

claell, 2019-09-28:
**Describe the bug**
I always get the message `Upgrade server down or could not connect` when searching for upgrades with `luci-app-attendedsysupgrade`.

I tried several different server URLs, none worked. Might be caused by https?

**Server URLs**
- https://chef.libremesh.org/
- https://as-test.stephen304.com/
- https://aparcar.stephen304.com/
- https://sudhanshu16.github.io/openwrt-firmware-selector/

**JSON request**
Unknown

**Build Log** 
Not relevant

**Expected behavior**
I expected to connect to a server.

**Screenshots**
[image]

Caveman8080, 2019-10-19:
I believe this is caused by CORS policy blocking when using https://aparcar.stephen304.com

Internet Explorer shows this error when searching for upgrades.
`Request header content-type was not present in the Access-Control-Allow-Headers list.`

Chrome show this error.
`Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource.`

claell, 2019-10-20:
It happened with every URL I tried (see the list above). Do they all have the same problem?

Caveman8080, 2019-10-20:
If I'm not mistaken, https://chef.libremesh.org/ no longer works, https://as-test.stephen304.com/ has the same error and I have not checked https://sudhanshu16.github.io/openwrt-firmware-selector/.

However, https://aparcar.stephen304.com/ should be the new address to use, but don't quote me on that.

BiatuAutMiahn, 2019-10-20:
Same issue, CORS

claell, 2019-10-21:
Alright. Ping @aparcar

tlaurion, 2019-11-12:
@aparcar ?
What is the complete list of servers that should be used?

tlaurion, 2019-11-14:
@aparcar :
Furthermore:
- https://chef.libremesh.org/ server doesn't provide images anymore from web gui and refers to aparcar.stephen304.com
- aparcar.stephen304.com provides snapshot only for rango (WRT3200ACM) and refers to https://sudhanshu16.github.io/openwrt-firmware-selector/ for new gui
- https://sudhanshu16.github.io/openwrt-firmware-selector/ doesn't provide images for rango, neither from snapshot or stable.

What is going on?

tlaurion, 2020-02-05:
@aparcar : is this project dead?

aparcar, 2020-02-25:
Ugh, so many URLs, sorry for the mess. I'm working on https://chef.libremesh.org/ which is a fast server I'm likely able to use for the future.

I have to re-implement the LuCi app which take a little while, however image building via the front-end should already work

closed 2020-02-25 by aparcar

CHazz, 2021-04-25:
checking https://chef.libremesh.org/ for release upgrade or updated packages
HTTP error 404
server reply invalid.

bam80, 2021-06-19:
ping?

Caveman8080, 2021-06-29:
If you are using OpenWRT 19.07 or earlier, you might not be able to use attended sysupgrade with https://chef.libremesh.org/ . I believe there was a change to the code and now the website is working with OpenWRT 21.02 a later.

aparcar, 2021-06-29:
The API changed at some point and I didn't backport the changes yet, sorry for the inconvenience. I'm on it.