openwrt/asu #203: Can't create images for Raspberry Pi 3B/3B+

code403, 2019-10-07:
Hello, i keep getting an error "An unexpected error occurred. Please try again" if i try to create images (18.06.4/Snapshot) for all Raspberry Pi devices

ich0x, 2019-10-07:
A bunch of targets are inactive on the imagebuilder at the moment, what is happening to this target brcm2708/* also is happening to the ramips/mt7621 grouping as well

Slluxx, 2020-01-11:
still an issue. this is probably the only way i can offline install luci on my raspi (as im in a dorm with very specific one-device-only access and bridge'ing doesnt work)

please fix

aparcar, 2020-02-24:
Works with the new server https://chef.libremesh.org/

[image]

closed 2020-02-24 by aparcar