openwrt/asu #205: Please add option to preselect compile and include as module or static selected hardware. Like in menuconfig for Kernel.

devegoo, 2019-10-30:
Please add option to preselect compile and include as module or static preselected dvb-usb dvb-usb V2 devices. This is my and other people who have USB port on router and DVB-T/T2 or DVB-S/S2 USB devices most wanted feature to have precompiled OpenWrt image with DVB funcionality.

devegoo, 2019-10-30:
Other way can be option to have only for stable OpenWrt ( for eg. 19.07)branches for devices/architectures option to download as module "*.ko" for folowed Kernel revisions in OpenWrt branch precompilef module needed for USB DVB dongle and coresponding additional firmware where needed.

aparcar, 2020-03-05:
Hi, you can easily install kmods and thereby USB drivers via the web interface, however compiling takes much to long, so no custom kernel options are changeable in the future, except you or anyone can provide a really fast server...

aparcar, 2021-03-14:
I'll close this as the user doesn't exists anymore anyway.

closed 2021-03-14 by aparcar