openwrt/asu #206: Zbtlink ZBT-APE522II unexpected error

jkimmel, 2019-11-07:
unexpected error comes up for Snapshot 19.07 and 18.06 .
Snapshot isn't available at all

aparcar, 2020-02-24:
Currently a problem of upstream... https://chef.libremesh.org/store/SNAPSHOT/ramips/mt7620/zbtlink_zbt-ape522ii/e3b0c44298fc/buildlog.txt

jkimmel, 2020-02-25:
> Currently a problem of upstream... https://chef.libremesh.org/store/SNAPSHOT/ramips/mt7620/zbtlink_zbt-ape522ii/e3b0c44298fc/buildlog.txt
> 
not working (404)

aparcar, 2020-02-25:
https://chef.libremesh.org/store/SNAPSHOT/ramips/mt7620/zbtlink_zbt-ape522ii/e3b0c44298fc/buildlog.txt

Changed the storage path - sorry...

aparcar, 2020-03-05:
Works again

[image]

closed 2020-03-05 by aparcar