openwrt/asu #207: help

chancat87, 2019-11-30:
Is there support for x86 and x64?

aparcar, 2020-02-24:
Currently not as the build process don't create any JSON files yet, which is required by the front end to know which images were actually created.

I'll try to fix this upstream but it might take a while - sorry

aparcar, 2020-03-19:
I created an upstream patch for that https://github.com/openwrt/openwrt/pull/2772

aparcar, 2020-03-22:
The patch is accepted and I have to integrate it into the build system.

Toomoch, 2020-04-14:
was support implemented?

FreifunkUFO, 2020-11-05:
on http://chef.libremesh.org/ now i can choose  Generic x86/64 on SNAPSHOT.
but, build throw an error
`opkg_install_cmd: Cannot install package procd.`
https://chef.libremesh.org/store/snapshot/x86/64/generic/e3b0c44298fc/buildlog.txt

aparcar, 2021-03-14:
Support for x86 packages was added some time ago.

aparcar, 2021-03-14:
[image]

closed 2021-03-14 by aparcar