openwrt/asu #209: unuspected error looking for Tplin-Archer C2600 images

desigabri, 2020-01-06:
Describe the bug
Obtaining ERROR MESSAGE: "unuspected error - Try again"  looking for Tplink-Archer C2600 images"

Server URL
https://sudhanshu16.github.io/openwrt-firmware-selector/

Screenshots
https://drive.google.com/open?id=1tXGyLQg_t6wWlNyj6w6L3TOWKtn3hOlb
https://drive.google.com/open?id=1_dLprNmKtpjICClen8E2qntsDazBei3u

aparcar, 2020-02-24:
Works on the new server https://chef.libremesh.org

[image]

desigabri, 2020-03-08:
yes it works via WEB, but it doesn't through the openwrt sysupgrade
attended luci-app  for automated package selection for the updated firmware
file.

Anyway I took note of this address THANKS

gabriele

aparcar, 2020-03-09:
in what way doesn't it work?

desigabri, 2020-03-09:
After your answer, I tried again to launch the attended sysupgrade command:

In my home router, it started requesting (first time it happened just
today) but still running (...maybe that server is compiling?)

In other 3 routers (always TP-link C2600), the page respond " Unsupported
profile: tplink,c2600"

gabriele

aparcar, 2020-03-09:
Should be fixed upstream now https://github.com/aparcar/asu/pull/230 Please also install the latest `luci-app-attendedsysupgrade`

desigabri, 2020-03-09:
Hi, thanks for your patience
 I searched for Attended sysupgrade updates but it seems to me to have the
latest one. I tried again and now I've again the first C2600 that is
working (the well is rotating but nothing happened) and  all the others 3
routers now tell me that there is a new snapshot but it fails to compile
(?).

 Is the server able to compile one firmware a time? Maybe that should it be
better if something in the page tell it to the users. something like:
"please wait for your turn: more requestes are in queue"

thanks

gabriele

[image: immagine.png]
[image: immagine.png]

desigabri, 2020-03-09:
this is my actual attendedsysupgrade version

luci-app-attendedsysupgrade    git-20.066.64757-78ec13b-1   ~6.5 KB    LuCI
support for attended sysupgrades

The nice message is " New release *snapshot* available - Installed version:
SNAPSHOT "

and the failure one is " AssertionError: ImageBuilder failed"

aparcar, 2020-03-09:
Yes that should be added, see the issue https://github.com/aparcar/asu/issues/222.

The server shows a lot of requests for different types of images, I need to implement some kind of debugging help that you can see the build log yourself :)

Please let me know what devices you're using and I see if I can figure out what fails.

desigabri, 2020-03-09:
At this time I want to compile Tp-link C2600 firmwares (2 differents
configurations). I'm used to compile them every some time by myself olso
for older devices, but those older devices won't be supported anymore
because of their minimal ROM and RAM and no space for attended sysupgrade
package  (TP-link TL-WR841, TL-MR3420, TL-WR1043, TEW-652 BRP), so I just
need firmwares for Archer C2600 v1, Archer C7 v5, TL-WR1043N v5 devices and
it should be really great if the page will work fine.

I think that this tool could be used to try packages and features directly
installing packages and after that, found want we need and what works and
do an Attended sysupgrade command to freeze a working firmware.

really thanks

gabriele

aparcar, 2021-08-29:
I'm closing this since the issue seem resolved

closed 2021-08-29 by aparcar