openwrt/asu #211: Is this project dead?

tlaurion, 2020-02-05:
https://github.com/aparcar/attendedsysupgrade-server/issues/201

FreifunkUFO, 2020-02-05:
yes, would be nice to know..
a new frontend is advertized somehow ( https://sudhanshu16.github.io/openwrt-firmware-selector/ ), but it has only fragments of former existing distributions and features :-/

all existing chef's are out of order, at least for ath-devices:
https://chef.libremesh.org/ -> no distribution available
https://chef.aparcar.org/ -> no distribution available
https://chef.aparcar.org/ -> no distribution available
https://aparcar.stephen304.com/ -> only one distribution / only for some ramips devices
https://sudhanshu16.github.io/openwrt-firmware-selector/ -> only for some ramips devices

aparcar, 2020-02-09:
Hey all, sorry for the bad situation & slow response time! The project isn't dead for me, I think my initial scope was just to big and implementation ideas just wrong. It ended up in something super complicated, way to bad coded and documented for anybody sane to join. However I still see the need of an easy mechanism to upgrade routers.

I'm thinking of "refactoring" the image creation and updates checks by something much simpler. I somewhat came up with using the [meta imagebuilder](https://github.com/aparcar/openwrt-metabuilder) with a Redis cache and a minimal queue to allow running builders in parallel.

A second step could be a simplified upgrade check, I [posted a proof of concept](https://github.com/openwrt/openwrt/pull/2366) some time ago. If OpenWrt would offer these JSON files the active ASU service would only handle builds -  no longer upgrade checks.

I'll create a minimal working version soon, combined with some upstream work on the JSON files.

If somebody is willing to work on that project with me, please join in and let's discuss details...

aparcar, 2020-02-21:
Here we go https://images.aparcar.org

New implementation based on python-rq (Redis Queue).

Please report errors. I'll merge the code into this repo once some docs & tests are added.

tlaurion, 2020-02-21:
@aparcar : sorry if I'm not following all intrications of #141 #133 but for the user trying to use luci-attended-sysupgrade, the result is still 405 errors and systems are still not upgradeable. Or am I missing something?

tlaurion, 2020-02-21:
Anything that should be said to test an upgrade? Cli based? Something?

aparcar, 2020-02-21:
Hi @tlaurion , the luci-app-sysupgrade needs to be upgraded and therefore can't be used just yet - sorry. Also CLI is not yet possible but I'll talk to @dangowrt who implemented the current version.

I focused on the frontend first to have something visible (also @mwarning was kind enough to help). I'll now prepare the merge and then continue to upgrade luci/cli apps.

aparcar, 2020-02-25:
I'd conclude this project is alive!

closed 2020-02-25 by aparcar