openwrt/asu #238: imagebuilder timeout

ghoffman9, 2020-03-20:
apacar -
your online imagebuilder is timing out for any new build this AM
https://chef.libremesh.org/

good morning and best wishes for health

aparcar, 2020-03-20:
Please retry

ghoffman9, 2020-03-20:
This is what I get:

Previously it took 180 sec to give this.

Now – on a different computer – it was immediate.

aparcar, 2020-03-20:
Hi, In case the servers from downloads.openwrt.org are to slow the build timeouts, I updated the timeout to 5 minutes and will try to use the faster CDN in the future. Please retry and close this issue if resolved.

ghoffman9, 2020-03-20:
Seems to work – great – thank you!

ghoffman9, 2020-03-30:
@aparcar -
another imagebuilder hangup.
for the past two days, it has ended with this error:
[image]
[imagebuildererror.txt](https://github.com/aparcar/asu/files/4402187/imagebuildererror.txt)

aparcar, 2020-03-31:
Seems to be an upstream issue and multiple devices are affected https://github.com/openwrt/openwrt/commit/2d61f8821c7cf99354e904139226c132554ba180

aparcar, 2020-03-31:
https://patchwork.ozlabs.org/patch/1264611/

ghoffman9, 2020-03-31:
Very good catch. Thank you

George

 

https://patchwork.ozlabs.org/patch/1264611/

aparcar, 2021-03-14:
Should be fixed.

closed 2021-03-14 by aparcar