openwrt/asu #256: Package size mismatch: ip-tiny is 122284 bytes, expecting 122262 bytes

FreifunkUFO, 2020-11-05:
trying to build 19.07.4 (ar71xx/generic) libremesh with this freifunk-profile from @dangowrt 
with these packages:
https://raw.githubusercontent.com/libremesh/network-profiles/389613e5e23560cb3488c777a54d46c129a34b42/freifunk/80211s-sae-bmx7-batadv/PACKAGES
and including
-dnsmasq -ppp

gives me 
` opkg_install_pkg: Package size mismatch: ip-tiny is 122284 bytes, expecting 122262 bytes`
https://chef.libremesh.org/store/19.07.4/ar71xx/generic/tl-wr1043nd-v3/e3b0c44298fc/buildlog.txt

aparcar, 2020-11-05:
Should be fixed via https://github.com/openwrt/openwrt/commit/fd29f2bcd97f7af7de22ac95b1490a54f45d6c88 which isn't backported

aparcar, 2020-11-05:
Problem is that packages contains the source date of `openwrt.git`, which constantly change while package releases may don't.

FreifunkUFO, 2020-11-05:
even more messages when trying SNAPSHOT
https://chef.libremesh.org/store/snapshot/ath79/generic/tplink_tl-wr1043nd-v3/e3b0c44298fc/buildlog.txt

aparcar, 2020-11-05:
Excellent, so this bug wasn't fixed. I'll give it another look.

FreifunkUFO, 2020-11-06:
ok thanks.
workaround for the ip-tiny-problem is to add to the PACkAges:
`ip-full -ip-tiny`
ps: so for now i have a running mesh with 19.07.4 and libremesh with freifunk-profile:
`profile-freifunk-80211s-sae-bmx7-batadv -dnsmasq ip-full -ip-tiny -wpad-basic luci-app-bmx7 luci-theme-bootstrap lime-hwd-openwrt-wan`

aparcar, 2021-03-14:
This issue was fixed in 19.07.7, 21.02.x and snapshots.

closed 2021-03-14 by aparcar