openwrt/asu #257: add freifunk-packages to the server

FreifunkUFO, 2020-11-05:
please add https://github.com/freifunk/openwrt-packages

aparcar, 2020-11-05:
Depends on backporting this PR:
https://github.com/openwrt/openwrt/commit/221f97ff4737f012c90feb086bc1c2ed86c6001b

FreifunkUFO, 2020-11-06:
but, you added libremesh-packages to asu. (why not the same way for freifunk-packages)
or, is the libremesh-repo added to openwrt yet?

aparcar, 2020-11-06:
Fair enough.

aparcar, 2020-11-06:
Okay I have to add an mechanism to add arch specific feeds, all LiMe packages are arch-unspecific. If you have some time for PRs please resolve https://github.com/freifunk/openwrt-packages/issues/29

aparcar, 2021-03-14:
There isn't much progress on adding the the architecture and I can't find any compiled packages anway, since it'd been removed from openwrt.git.

SvenRoederer, 2021-05-13:
So you need binary-packages in the usual feed-hierarchy for download?

regarding the ARCH-specific version I still waiting for a review of https://github.com/openwrt/luci/pull/2637 to decide on the direction of the next steps.

FreifunkUFO, 2021-05-21:
when i tried an openwrt from openwrt-website there is a repo for freifunk included:

f.i. https://downloads.openwrt.org/releases/19.07.7/packages/x86_64/freifunk/

FreifunkUFO, 2021-05-21:
so you have to add "freifunk" to the lists in these lines:
https://github.com/aparcar/asu/blob/master/misc/config.py#L34
https://github.com/aparcar/asu/blob/master/misc/config.py#L121
https://github.com/aparcar/asu/blob/master/misc/config.py#L213

aparcar, 2021-05-21:
Freifunk was removed from 21.02 and snapshots, is there another source which is maintained by Freifunk directly?

SvenRoederer, 2021-05-21:
Currently I'm not aware of any. If there is a real interest we can implements a github-workflow which is building via SDK. Sadly github is not offering flat file-storage, but I'm sure Freifunk can offer some Server-space to upload the files to.

FreifunkUFO, 2021-05-25:
omg, @aparcar so you could add these lines to this asu at least for openwrt 17.09!?
of course there is real interest, but i guess "some" freifunk is not able to offer proper freifunk-packages for all openwrt (webspace is the smallest problem).

btw libremesh has the same problem, there are no built packages to use with openwrt in a simple way (repo is https://github.com/libremesh/lime-packages ) 
so ASU has configured these libremesh-packages (not documented) and link repo to an undocumented, not accessible URL on the internet  ("lime-packages": "https://feed.libremesh.org/master",)

p.s. would be nice to see (and able to use) freifunk- and libremesh-packages on openwrt (again). at the moment f.i. the openwrt-imagebuilder is useless for creating meshcommunity images. also installing freifunk/libremesh-packages on an naked openwrt is crazy by that. but its handled and discussed somewhere else i guess...

aparcar, 2021-05-25:
Sorry for the inconvenience. I've created a simple *build and upload* [GitHub action](https://github.com/libremesh/lime-packages/blob/master/.github/workflows/build.yml) which uploads all LibreMesh packages to [libremesh/lime-feed](https://github.com/libremesh/lime-feed/). LibreMesh people (or me? don't remember) created a CNAME for that - sorry it's somewhat hidden.

I'd suggest we add the same for freifunk-packages and then I'll at it to ASU too.

aparcar, 2021-05-25:
I re-added the freifunk feed to 19.07, it was lost in a copy & paste action...

FreifunkUFO, 2021-05-27:
ok thank you. but i can't find it on https://github.com/aparcar/asu/blob/master/misc/config.py#L213

and, for 19-07 there are x86 targets in this config, but https://chef.libremesh.org/ doesnt show x86 possibility.

aparcar, 2021-05-27:
> ok thank you. but i can't find it on https://github.com/aparcar/asu/blob/master/misc/config.py#L213

I changed it on the running server, I'll push some updates soon including that patch. If you feel compelled to do so please create a PR.

> and, for 19-07 there are x86 targets in this config, but https://chef.libremesh.org/ doesnt show x86 possibility.

19.07 uses old image code for the x86 target and therefore doesn't produce JSON files, which is however crucial to work with ASU and the firmware selector.

FreifunkUFO, 2021-05-27:
ok, PR is here: https://github.com/aparcar/asu/pull/279

closed 2021-06-03 by FreifunkUFO