openwrt/asu #258: Package size mismatch for a lot of packages

tlaurion, 2020-12-03:
https://chef.libremesh.org/store/19.07.4/mvebu/cortexa9/linksys_wrt3200acm/e3b0c44298fc/buildlog.txt
```
### STDOUT

Building images for mvebu - Linksys WRT3200ACM (Rango)
Packages: luci-mod-status luci-base luci-mod-network luci-mod-admin-full luci-lib-jsonc luci-app-vpn-policy-routing luci-lib-iptparser luci-app-advanced-reboot luci-compat luci-app-opkg luci luci-lib-nixio luci-lib-ip luci-app-attendedsysupgrade luci-app-statistics luci-app-firewall base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables iwinfo kernel kmod-btmrvl kmod-gpio-button-hotplug kmod-ipt-offload kmod-mwifiex-sdio kmod-mwlwifi libc libgcc logd mtd mwlwifi-firmware-88w8964 netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic

Installing packages...
Installing libc (1.1.24-2) to root...
Installing libgcc1 (7.5.0-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libgcc1_7.5.0-2_arm_cortex-a9_vfpv3-d16.ipk.
Configuring libgcc1.
Configuring libc.
Installing kernel (4.14.195-1-a92a3f5c5bed2671533484c7ace9d5b5) to root...
Configuring kernel.
Installing luci-mod-status (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-status_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing liblua5.1.5 (5.1.5-3) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/liblua5.1.5_5.1.5-3_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-base (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-base_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-mod-network (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-network_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libiwinfo-lua (2019-10-16-07315b6f-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libiwinfo-lua_2019-10-16-07315b6f-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing libubus20191227 (2019-12-27-041c9d1c-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libubus20191227_2019-12-27-041c9d1c-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-mod-admin-full (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-admin-full_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-mod-status (git-20.334.47525-e9bfb12-1) to root...
Installing luci-mod-system (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-system_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-mod-network (git-20.334.47525-e9bfb12-1) to root...
Installing luci-lib-jsonc (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-jsonc_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-app-vpn-policy-routing (git-20.334.47525-e9bfb12-74) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-vpn-policy-routing_git-20.334.47525-e9bfb12-74_all.ipk.
Installing luci-compat (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-compat_git-20.334.47525-e9bfb12-1_all.ipk.
Installing kmod-nf-reject (4.14.195-1) to root...
Downloading file:packages/kmod-nf-reject_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-nf-ipt (4.14.195-1) to root...
Downloading file:packages/kmod-nf-ipt_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-ipt-core (4.14.195-1) to root...
Downloading file:packages/kmod-ipt-core_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-nfnetlink (4.14.195-1) to root...
Downloading file:packages/kmod-nfnetlink_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-ipt-ipset (4.14.195-1) to root...
Downloading file:packages/kmod-ipt-ipset_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing libmnl0 (1.0.4-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libmnl0_1.0.4-2_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-lib-iptparser (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-iptparser_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-app-advanced-reboot (git-20.334.47525-e9bfb12-54) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-advanced-reboot_git-20.334.47525-e9bfb12-54_all.ipk.
Package luci-compat (git-20.334.47525-e9bfb12-1) installed in root is up to date.
Installing luci-app-opkg (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-opkg_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libubox20191228 (2020-05-25-66195aee-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libubox20191228_2020-05-25-66195aee-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libubox20191228 (2020-05-25-66195aee-1) to root...
Installing luci-lib-nixio (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-nixio_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-lib-ip (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-ip_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-app-attendedsysupgrade (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-attendedsysupgrade_git-20.334.47525-e9bfb12-1_all.ipk.
Installing uhttpd (2020-10-01-3abcc891-1) to root...
[... 246 lines cut ...]
 * opkg_install_pkg: Package size mismatch: odhcpd-ipv6only is 34641 bytes, expecting 34627 bytes
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
 * opkg_install_pkg: Package size mismatch: opkg is 67308 bytes, expecting 67302 bytes
 * opkg_install_cmd: Cannot install package opkg.
 * opkg_install_pkg: Package size mismatch: ppp is 111762 bytes, expecting 111752 bytes
 * opkg_install_cmd: Cannot install package ppp.
 * opkg_install_pkg: Package size mismatch: ppp-mod-pppoe is 10765 bytes, expecting 10764 bytes
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * opkg_install_pkg: Package size mismatch: swconfig is 9339 bytes, expecting 9345 bytes
 * opkg_install_cmd: Cannot install package swconfig.
 * opkg_install_pkg: Package size mismatch: uci is 7471 bytes, expecting 7473 bytes
 * opkg_install_cmd: Cannot install package uci.
 * opkg_install_pkg: Package size mismatch: uclient-fetch is 8034 bytes, expecting 8046 bytes
 * opkg_install_cmd: Cannot install package uclient-fetch.
 * opkg_install_pkg: Package size mismatch: urandom-seed is 1548 bytes, expecting 1549 bytes
 * opkg_install_cmd: Cannot install package urandom-seed.
 * opkg_install_pkg: Package urngd sha256sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package urngd.
 * opkg_install_pkg: Package size mismatch: wpad-basic is 333594 bytes, expecting 333583 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [Makefile:158: package_install] Error 255
make[1]: *** [Makefile:118: _call_image] Error 2
make: *** [Makefile:208: image] Error 2
```
@aparcar

tlaurion, 2020-12-03:
Without any additionally selected packages.
https://chef.libremesh.org/store/19.07.4/mvebu/cortexa9/linksys_wrt3200acm/e3b0c44298fc/buildlog.txt
```
### STDOUT

Building images for mvebu - Linksys WRT3200ACM (Rango)
Packages: luci-mod-status luci-base luci-mod-network luci-mod-admin-full luci-lib-jsonc luci-app-vpn-policy-routing luci-lib-iptparser luci-app-advanced-reboot luci-compat luci-app-opkg luci luci-lib-nixio luci-lib-ip luci-app-attendedsysupgrade luci-app-statistics luci-app-firewall base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables iwinfo kernel kmod-btmrvl kmod-gpio-button-hotplug kmod-ipt-offload kmod-mwifiex-sdio kmod-mwlwifi libc libgcc logd mtd mwlwifi-firmware-88w8964 netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic

Installing packages...
Installing libc (1.1.24-2) to root...
Installing libgcc1 (7.5.0-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libgcc1_7.5.0-2_arm_cortex-a9_vfpv3-d16.ipk.
Configuring libgcc1.
Configuring libc.
Installing kernel (4.14.195-1-a92a3f5c5bed2671533484c7ace9d5b5) to root...
Configuring kernel.
Installing luci-mod-status (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-status_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing liblua5.1.5 (5.1.5-3) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/liblua5.1.5_5.1.5-3_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-base (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-base_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-mod-network (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-network_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libiwinfo-lua (2019-10-16-07315b6f-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libiwinfo-lua_2019-10-16-07315b6f-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing libubus20191227 (2019-12-27-041c9d1c-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libubus20191227_2019-12-27-041c9d1c-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-mod-admin-full (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-admin-full_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-mod-status (git-20.334.47525-e9bfb12-1) to root...
Installing luci-mod-system (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-mod-system_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-mod-network (git-20.334.47525-e9bfb12-1) to root...
Installing luci-lib-jsonc (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-jsonc_git-20.334.47525-e9bfb12-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-app-vpn-policy-routing (git-20.334.47525-e9bfb12-74) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-vpn-policy-routing_git-20.334.47525-e9bfb12-74_all.ipk.
Installing luci-compat (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-compat_git-20.334.47525-e9bfb12-1_all.ipk.
Installing kmod-nf-reject (4.14.195-1) to root...
Downloading file:packages/kmod-nf-reject_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-nf-ipt (4.14.195-1) to root...
Downloading file:packages/kmod-nf-ipt_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-ipt-core (4.14.195-1) to root...
Downloading file:packages/kmod-ipt-core_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-nfnetlink (4.14.195-1) to root...
Downloading file:packages/kmod-nfnetlink_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing kmod-ipt-ipset (4.14.195-1) to root...
Downloading file:packages/kmod-ipt-ipset_4.14.195-1_arm_cortex-a9_vfpv3-d16.ipk
Installing libmnl0 (1.0.4-2) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libmnl0_1.0.4-2_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci-lib-iptparser (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-lib-iptparser_git-20.334.47525-e9bfb12-1_all.ipk.
Installing luci-app-advanced-reboot (git-20.334.47525-e9bfb12-54) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-advanced-reboot_git-20.334.47525-e9bfb12-54_all.ipk.
Package luci-compat (git-20.334.47525-e9bfb12-1) installed in root is up to date.
Installing luci-app-opkg (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci-app-opkg_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libubox20191228 (2020-05-25-66195aee-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/libubox20191228_2020-05-25-66195aee-1_arm_cortex-a9_vfpv3-d16.ipk.
Installing luci (git-20.334.47525-e9bfb12-1) to root...
Copying /home/aparcar/asu/var/asu-instance/cache/19.07.4/mvebu/cortexa9/dl/luci_git-20.334.47525-e9bfb12-1_all.ipk.
Installing libubox20191228 (2020-05-25-66195aee-1) to root...
[... cut ...]
 * opkg_install_cmd: Cannot install package urandom-seed.
 * opkg_install_pkg: Package urngd sha256sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package urngd.
 * opkg_install_pkg: Package size mismatch: wpad-basic is 333594 bytes, expecting 333583 bytes
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [Makefile:158: package_install] Error 255
make[1]: *** [Makefile:118: _call_image] Error 2
make: *** [Makefile:208: image] Error 2

```

tlaurion, 2021-01-04:
@aparcar ping

tlaurion, 2021-01-05:
https://chef.libremesh.org/store/19.07.4/mvebu/cortexa9/linksys_wrt3200acm/e3b0c44298fc/buildlog.txt

tlaurion, 2021-01-05:
where 19.07.5 produces no result and where SNAPSHOT produces a recoverale brick because dual flash (with trick of rebooting 3 times to reboot in alternate rom)

aparcar, 2021-01-05:
I'm somewhat caught up in upstream things I try to improve but I'll try to get some improvements going within this week.

aparcar, 2021-03-14:
This should be fixed in 21.02 and snapshots and even the latest 19.07.7 release. Please open again if the error still occurs.

closed 2021-03-14 by aparcar