openwrt/asu #259: Extroot Firmware Configuration

NovaViper, 2020-12-29:
Hey, does the luci app have support for generating firmware configs that have the necessary packages for routers with extroot? I have an Archer A7 V5 with a 8GB USB stick connected to it and I have my packages loaded onto that via Extroot.

aparcar, 2021-03-14:
Good questions, from my understanding you have some packages "locally" installed and some "externally"? Do all these packages show up the same way when you run `opkg list-installed`?

If so, the current implementation can not tell what package was locally or externally installed.

NovaViper, 2021-03-14:
I actually can't verify if the packages are on the machine's chips themselves or the usb because I scrapped the configuration last week. But from what I remember the packages were all moved over to the usb (hence, externally mounted)

erdoukki, 2021-08-14:
May be opkg extras will be usefull for the autoupgrade/autorestore process : https://openwrt.org/docs/guide-user/advanced/opkg_extras

ortegafernando, 2022-08-28:
Hi, does anybody know if asu works with extroot ? Thanks