openwrt/asu #260: Step by step setup guide

Caveman8080, 2021-01-05:
Is there a good step by step guide I could use to setup ASU? I'm not good with all of this stuff, and every time I try to set it up, I run into issues like the builder not building, python returning errors or the server giving back 400 errors.

My end goal is to create a single Docker YAML file that I can upload to my QNAP NAS and run ASU via container station.

agusalex, 2021-01-25:
I wanted to chime in on this, any reason why Docker and docker-compose is not supported anymore? awesome work btw

aparcar, 2021-04-13:
Hi, I extended the CI a bit and now there are always [fresh Docker containers](https://hub.docker.com/r/aparcar/asu/tags?page=1&ordering=last_updated) and the installation via [PyPi](https://pypi.org/project/asu/#history) should also result in more recent versions.

For a *to be created* `docker-compose.yml` file the following services should run:
* A single `asu` service
* A `redis` server
* A service which updates metadata (profiles, packages, versions)
  * I used to have a `cronjob` which is ugly, [this](https://github.com/aparcar/asu/pull/242) needs some rebasing to run as a single server
* Any number of `workers` which should be based on the Docker container `openwrt/imagebuilder:latest` and have `rqworker` installed.

aparcar, 2021-08-11:
I updated the Docker stuff, please report your results. Ideally there is a Caddy/Nginx container in front of it, PRs welcome.

https://github.com/aparcar/asu#docker

closed 2021-08-11 by aparcar

bam80, 2022-09-24:
There still no guide, right?
It's not clear for novice user how to set up his own ASU instance serving custom-built packages.
It's not  even known to me what `rqworker` is.
I probably don't know some basics, so refs to the documentation required should be at least included, I think.

It's not complaint but rather a wish to make the project even better.
Thanks.