openwrt/asu #264: TP Link Archer MR200 v4 support needed

Mctifa, 2021-02-09:
Please add the TP Link Archer MR200 v4 firmware to the list as I cannot find it.

aparcar, 2021-02-10:
It's available, I accidentally messed something up in the fronted. Please retry.

[image]

closed 2021-02-10 by aparcar

Mctifa, 2021-02-10:
@aparcar please note that this is not for v4

aparcar, 2021-02-10:
Is the device support by OpenWrt? If not, the server can't build the image.

aparcar, 2021-02-10:
This commit never made it into upstream OpenWrt: https://github.com/Mctifa/mr200v4/commit/87418bafc8fd1151eed32d265f956d168c50efc1

Mctifa, 2021-02-10:
Thank you for the heads up
I will try to build it and see

asaorgad, 2021-07-07:
Hi , so from all the discussion above, i didnt understood, is there a support for openwrt for mr200 v4 ? If yes can someone pleae send me the link to the firmware .?

aparcar, 2021-07-07:
@asaorgad it's not part of OpenWrt, you could ask @Mctifa to send the link to upstream.