openwrt/asu #267: get no image from chef if i add profile-freifunk-80211s-sae-bmx7-batadv

der-nele, 2021-03-17:
if i just put this  -libustream-wolfssl -wpad-basic-wolfssl libustream-openssl openssl-util wpad-openssl 
it works

https://chef.libremesh.org/?version=21.02-SNAPSHOT&target=ramips%2Fmt7621&id=tplink_re650-v1
the same whith  linksys / belkin RT3200 

by the way THANKS for this work

asu.build.PackageSelectionError

log

Signature check passed.
Downloading https://feed.libremesh.org/master/Packages.gz
Updated list of available packages in /home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../../home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/dl/lime-packages
Downloading https://feed.libremesh.org/master/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://feed.libremesh.org/profiles/Packages.gz
Updated list of available packages in /home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../../home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/dl/lime-profiles
Downloading https://feed.libremesh.org/profiles/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package libgcc1. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package libc.
make[2]: *** [Makefile:164: package_install] Error 255
make[1]: *** [Makefile:131: _call_manifest] Error 2
make: *** [Makefile:250: manifest] Error 2

aparcar, 2021-03-17:
This is an issue with LibreMesh package and not of ASU itself, waiting for "upstream" to fix that.

der-nele, 2021-03-18:
ok thanks than i wait too

der-nele, 2021-03-27:
i see there is going same things in libremesh. just thought 
maybe there can be an option in chef like -F profile-freifunk-80211s-sae-bmx7-batadv  (force pakate or ignore key) to be more independent.

der-nele, 2021-04-08:
ok now there is  mistake is different.
this is now an profile problem? or is there a possebilaty to handel this in chef? 

Downloading https://feed.libremesh.org/profiles/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package wpad-mesh-wolfssl wants to install file /home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/hostapd
	But that file is already provided by package  * wpad-basic-wolfssl
 * check_data_file_clashes: Package wpad-mesh-wolfssl wants to install file /home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpa_supplicant
	But that file is already provided by package  * wpad-basic-wolfssl
 * check_data_file_clashes: Package wpad-mesh-wolfssl wants to install file /home/aparcar/asu/var/asu-instance/cache/21.02-SNAPSHOT/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpad
	But that file is already provided by package  * wpad-basic-wolfssl
 * opkg_install_cmd: Cannot install package profile-freifunk-80211s-sae-bmx7-batadv.
make[2]: *** [Makefile:167: package_install] Error 255
make[1]: *** [Makefile:132: _call_manifest] Error 2
make: *** [Makefile:251: manifest] Error 2

aparcar, 2021-04-08:
The following package selection seem to solve the issue. Fixing dependency issues within LibreMesh is not my interest anymore...

```
base-files busybox ca-bundle dropbear firewall fstools ip6tables iptables kmod-ath9k kmod-gpio-button-hotplug kmod-ipt-offload kmod-usb-ledtrig-usbport kmod-usb2 libgcc libustream-wolfssl logd luci mtd netifd odhcp6c opkg ppp ppp-mod-pppoe procd swconfig uboot-envtools ucert uci uclient-fetch urandom-seed urngd wpad-mesh-wolfssl profile-freifunk-80211s-sae-bmx7-batadv
```

der-nele, 2021-04-10:
ok thanks for help! but the firmware crasht my node ;(  maybe there are more skills needet for using this.

aparcar, 2021-04-10:
Sorry no idea what happened. If the issue is not a wrong package selection i don't know how to help.

der-nele, 2021-04-10:
jes i just wontet to say that this dosent work. i have no idear why not but i get some help next week. then i try it again and give a feedback i you wont.

der-nele, 2021-04-16:
ok so the router is fine. just do not come up whith interfaces which is an snapshot problem.
or problem of the not full seproted hartware i use.
thanks a lot

closed 2021-04-16 by der-nele