openwrt/asu #277: version in 19.07.7 (latest stable) not working ?

CHazz, 2021-04-25:
hi i try for check few times in many versions of openwrt but never work ..
now have 19.07.7 and again nothing ..  no way 

`root@OpenWrt:~# auc
auc (0.1.4)
running OpenWrt 19.07.7 on ath79/generic (tplink,archer-c7-v5)
checking https://chef.libremesh.org/ for release upgrade or updated packages
HTTP error 404
server reply invalid.
`

aparcar, 2021-04-26:
I'm sorry for the inconvenience, @dangowrt and me did some efforts to make the experience much better in version 21.02 an onward. Please try the latest 21.02.0-rc1

richb-hanover, 2021-04-27:
I also saw this error message (Netgear WNDR3800 running 19.07.7). Is it the same situation? (I am not asking for any change - I'm totally fine with holding the ASU until we ship 21.02...)

[image]

CHazz, 2021-04-30:
in 21.02.0-rc1 working :) Thank you and Good Luck

CHazz, 2021-05-01:
i try flast to snapshot .. working . is possible flash bact to future stable version from luci ?

richb-hanover, 2021-08-16:
I get a permanent "Searching for an available sysupgrade" hang in the LuCI GUI... (I'm using 19.07.7 on Linksys WRT3200ACM).

Is this the same symptom? Should I just upgrade to 21.02 manually? Thanks.

[image]

aparcar, 2021-08-29:
@richb-hanover can you give an update on this? still facing the issue?

richb-hanover, 2021-08-29:
Solved. I went to the router (running 19.07.7) and clicked "Search for sysupgrade"

I got a spinner for a couple seconds (I think) then was presented with the option to upgrade to 19.07.8. There were no other choices in the dropdown. The router is in production now, so I did not try the upgrade. I will try it next weekend.

Thanks!

aparcar, 2021-08-29:
Please let me know how it goes.

aparcar, 2021-09-24:
Updates on this?

richb-hanover, 2021-09-25:
Alas... I have been extremely busy and the site where that router is installed has been in production with lots of users. I did go to the GUI and it offered to build and then download, but I (again) decided not to do it. Perhaps you could close this (the original problem I reported seems fixed) and I will re-open if I see a problem when I actually upgrade. Thanks.

richb-hanover, 2021-09-28:
@aparcar Using luci-app-attendedsysupgrade git-21.267.59215-1524972-1 on OpenWrt 19.07.7 on Linksys WRT3200ACM. LuCI says "Powered by LuCI openwrt-19.07 branch (git-21.044.30835-34e0d65) / OpenWrt 19.07.7 r11306-c4a6851c72" at the bottom of each page.

I went to **System -> Attended Sysupgrade** and was given the choice of 19.07.8. I clicked "Request sysupgrade" and immediately got the following error window:

**Error building the sysupgrade**
Unsupported package(s): libubus20191227
**Please report the error message and request**

Request to server:
```
Request to server:
{
"board_name": "linksys,rango",
"target": "mvebu/cortexa9",
"version": "19.07.8",
"packages": [
"base-files",
"busybox",
"cgi-io",
"dnsmasq",
"dropbear",
"firewall",
"fstools",
"fwtool",
"getrandom",
"ip6tables",
"iptables",
"iwinfo",
"jshn",
"jsonfilter",
"kernel",
"kmod-btmrvl",
"kmod-gpio-button-hotplug",
"kmod-ip6tables",
"kmod-ipt-conntrack",
"kmod-ipt-core",
"kmod-ipt-nat",
"kmod-ipt-offload",
"kmod-lib-crc-ccitt",
"kmod-mwifiex-sdio",
"kmod-mwlwifi",
"kmod-nf-conntrack",
"kmod-nf-conntrack6",
"kmod-nf-flow",
"kmod-nf-ipt",
"kmod-nf-ipt6",
"kmod-nf-nat",
"kmod-nf-reject",
"kmod-nf-reject6",
"kmod-ppp",
"kmod-pppoe",
"kmod-pppox",
"kmod-slhc",
"libblobmsg-json",
"libip4tc2",
"libip6tc2",
"libiwinfo-lua",
"libiwinfo20181126",
"libjson-c2",
"libjson-script",
"liblua5.1.5",
"liblucihttp-lua",
"liblucihttp0",
"libnl-tiny",
"libubox20191228",
"libubus-lua",
"libubus20191227",
"libuci20130104",
"libuclient20160123",
"libxtables12",
"logd",
"lua",
"luci",
"luci-app-attendedsysupgrade",
"luci-app-firewall",
"luci-app-opkg",
"luci-app-snmpd",
"luci-app-sqm",
"luci-app-wireguard",
"luci-base",
"luci-lib-ip",
"luci-lib-jsonc",
"luci-lib-nixio",
"luci-mod-admin-full",
"luci-mod-network",
"luci-mod-status",
"luci-mod-system",
"luci-proto-ipv6",
"luci-proto-ppp",
"luci-theme-bootstrap",
"mtd",
"mwlwifi-firmware-88w8964",
"nano",
"netifd",
"odhcp6c",
"odhcpd-ipv6only",
"openwrt-keyring",
"opkg",
"ppp",
"ppp-mod-pppoe",
"procd",
"rpcd",
"rpcd-mod-file",
"rpcd-mod-iwinfo",
"rpcd-mod-luci",
"rpcd-mod-rrdns",
"snmpd",
"swconfig",
"ubi-utils",
"uboot-envtools",
"ubox",
"ubus",
"ubusd",
"uci",
"uclient-fetch",
"uhttpd",
"urandom-seed",
"urngd",
"usign",
"wireguard",
"wpad-basic"
],
".anonymous": false,
".name": "request",
".type": "request",
".index": 0
}
```

aparcar, 2021-09-28:
Which server are you using?

richb-hanover, 2021-09-28:
I changed the config tab to use `https://asu.aparcar.org` (it was chef.libremesh.org), and got the same error.
 
It's complaining about `libubus20191227` - is that a problem?

richb-hanover, 2021-09-28:
**More info:** I uninstalled `luci-app-attendedsysupgrade` and went to re-install it. I got this list of packages that would be installed:

[image]

and here's the installation report:

[image]

Re-attempting the attended sysupgrade gets the same error message.

aparcar, 2021-09-28:
What is the content of /etc/config/attendedsysupgrade

richb-hanover, 2021-09-29:
Contents of /etc/config/attendedsysupgrade

```
config server 'server'
	option url 'https://asu.aparcar.org'

config client 'client'
	option upgrade_packages '1'
	option auto_search '0'
	option advanced_mode '0'
```

aparcar, 2021-09-29:
Fixed via https://github.com/aparcar/asu/commit/bfecd216f362766cde3dfaa137fb0043e0c8bb79

aparcar, 2021-09-29:
Please start using `https://sysupgrade.openwrt.org` as upgrade server.

closed 2021-09-29 by aparcar

richb-hanover, 2021-09-29:
That new upgrade server seems to have worked... I did not actually perform the upgrade since the router is in production. I'll update if I have problems. Thanks!

[image]