openwrt/asu #278: Error building the sysupgrade: OpenWrt 21.02-rc1; WNDR3800

richb-hanover, 2021-04-27:
I installed OpenWrt 21.02-rc1 (Powered by LuCI openwrt-21.02 branch (git-21.106.55967-06dd6b5) / OpenWrt 21.02.0-rc1 r16046-59980f7aaf ) on a Netgear WNDR3800 router. I then installed the *luci-app-attendedsysupgrade* package and rebooted. 

I went to the System -> Attended Sysupgrade page and clicked *Search for sysupgrade*. I chose SNAPSHOT from the resulting window (it was the only choice) and got the following message:

> **Error building the sysupgrade**
> Missing target
> Please report the error message and request
> Request to server:
> ```
> {
>     "board_name": "netgear,wndr3800",
>     "version": "SNAPSHOT",
>     "packages": [
>         "base-files",
>         "busybox",
>         "ca-bundle",
>         "cgi-io",
>         "dnsmasq",
>         "dropbear",
>         "firewall",
>         "fstools",
>         "fwtool",
>         "getrandom",
>         "hostapd-common",
>         "ip6tables",
>         "iptables",
>         "iw",
>         "iwinfo",
>         "jshn",
>         "jsonfilter",
>         "kernel",
>         "kmod-ath",
>         "kmod-ath9k",
>         "kmod-ath9k-common",
>         "kmod-cfg80211",
>         "kmod-gpio-button-hotplug",
>         "kmod-ip6tables",
>         "kmod-ipt-conntrack",
>         "kmod-ipt-core",
>         "kmod-ipt-nat",
>         "kmod-ipt-offload",
>         "kmod-leds-reset",
>         "kmod-lib-crc-ccitt",
>         "kmod-mac80211",
>         "kmod-nf-conntrack",
>         "kmod-nf-conntrack6",
>         "kmod-nf-flow",
>         "kmod-nf-ipt",
>         "kmod-nf-ipt6",
>         "kmod-nf-nat",
>         "kmod-nf-reject",
>         "kmod-nf-reject6",
>         "kmod-owl-loader",
>         "kmod-ppp",
>         "kmod-pppoe",
>         "kmod-pppox",
>         "kmod-slhc",
>         "kmod-usb-ledtrig-usbport",
>         "kmod-usb-ohci",
>         "kmod-usb2",
>         "libblobmsg-json",
>         "libip4tc",
>         "libip6tc",
>         "libiwinfo",
>         "libiwinfo-lua",
>         "libjson-c",
>         "libjson-script",
>         "liblua",
>         "liblucihttp",
>         "liblucihttp-lua",
>         "libnl-tiny",
>         "libubox",
>         "libubus",
>         "libubus-lua",
>         "libuci",
>         "libuclient",
>         "libustream-wolfssl",
>         "libwolfssl",
>         "libxtables",
>         "logd",
>         "lua",
>         "luci",
>         "luci-app-attendedsysupgrade",
>         "luci-app-firewall",
>         "luci-app-opkg",
>         "luci-base",
>         "luci-lib-base",
>         "luci-lib-ip",
>         "luci-lib-jsonc",
>         "luci-lib-nixio",
>         "luci-mod-admin-full",
>         "luci-mod-network",
>         "luci-mod-status",
>         "luci-mod-system",
>         "luci-proto-ipv6",
>         "luci-proto-ppp",
>         "luci-theme-bootstrap",
>         "mtd",
>         "netifd",
>         "odhcp6c",
>         "odhcpd-ipv6only",
>         "openwrt-keyring",
>         "opkg",
>         "ppp",
>         "ppp-mod-pppoe",
>         "procd",
>         "rpcd",
>         "rpcd-mod-file",
>         "rpcd-mod-iwinfo",
>         "rpcd-mod-luci",
>         "rpcd-mod-rrdns",
>         "swconfig",
>         "uboot-envtools",
>         "ubox",
>         "ubus",
>         "ubusd",
>         "uci",
>         "uclient-fetch",
>         "uhttpd",
>         "uhttpd-mod-ubus",
>         "urandom-seed",
>         "urngd",
>         "usign",
>         "wireless-regdb",
>         "wpad-basic-wolfssl"
>     ],
>     ".anonymous": false,
>     ".name": "request",
>     ".type": "request",
>     ".index": 0
> }
> ```

aparcar, 2021-04-28:
https://github.com/openwrt/luci/pull/5014
Fix backported

richb-hanover, 2021-04-28:
Thanks! How do I test it?  I currently have uci-app-attendedsysupgrade | git-21.077.73862-d7568b4 installed...

aparcar, 2021-04-28:
Please wait for the recompilation and upgrade the app via opkg

richb-hanover, 2021-04-29:
Perfect! I installed luci-app-attendedsysupgrade (git-21.118.41464-33aa61f) this morning. It "acted right", and 25 seconds later, I had a new build of SNAPSHOT. 

[Here's a movie of it in action!](https://imgur.com/gallery/OHSvZgD)

I did not test the install (I have other projects today), but I wanted to confirm that the LuCI GUI for sysupgrade was working as expected. Thanks!

closed 2021-04-29 by aparcar