openwrt/asu #282: Feature Request - UCI Defaults

fonix232, 2021-06-09:
One of my main hurdles with OpenWRT is that whenever I mess up my configuration, everything is reset - I need a wired terminal to set things up, since WiFi isn't brought up by default, then manually pull in the working configuration (either via backup, or actually manually executing UCI commands).

OpenWRT solved this with a great tool - [UCI defaults](https://openwrt.org/docs/guide-developer/uci-defaults). They're incredibly easy to use, you just drop appropriately named files into `/etc/uci-defaults` and on every boot, these scripts are executed. The main issue? One either needs to bake these files into the firmware, or into a custom IPK package, for them to survive a reset. 

Since ASU already offers the possibility to bake packages into the firmware produced, I think it would be a great addition if a collection of custom configurations could be attached to the build request to create updates that come pre-configured for said device - meaning even in case of a factory reset, the user-set default configuration is applied on first boot.

I do understand that this can be extremely dangerous, since a bad `uci-default` configuration could render the device unusable (however this is true for generic ASU updates too - say, there's an issue in the latest SNAPSHOT version affecting a single device). At this point I have not thought of any possible safety measure to prevent this, nor did I think of possibilities of the configuration being caught during the request (which could lead to an attacker learning the user's full configuration, thereby providing easier access to the network). In fact I'd like to leave the topics of security and safety up for discussion.

I am also open to alternative solutions, e.g. packaging an IPK locally on the router, and sending that to the ASU server instead of cleartext configuration. Obviously the safest option would be to add the extra `uci-defaults` files to the image locally, after the "public" part of the image is built.

aparcar, 2021-08-05:
I used to support this feature but the issue is a bit with privacy, you end up having all your stuff configured with passwords etc hanging out on *my* server. Not sure if that's a proper solution.

fonix232, 2021-08-12:
@aparcar yes, I understand that part. An alternative solution could be supporting it only on local APIs (so the `luci-app` would have the option, but disabled for public facing servers like chef.libremesh.org), or, alternatively, compiling a set of tools + scripts that can be run on the local device to inject those bits and pieces. It's a pity OpenWrt's image format doesn't support append-style configuration.

Your recent change, however (removing builds after 15 minutes) should make things safer. Possibly even limiting the download to the IP where the build request came from. Maybe even a signal from the client for a successful download that deletes the server-side file immediately?

aparcar, 2021-08-13:
Implemented via https://github.com/aparcar/asu/commit/0f412dd48c271b65bcfdf3c6ebca13c2feda97e1 I'll test this a bit and merge it. It'll need modifications of the web interface which I'm not really willing to do...

aparcar, 2021-09-24:
This feature is enabled for asu.aparcar.org.

closed 2021-09-24 by aparcar