openwrt/asu #283: Upgrade server down or could not connect

bam80, 2021-06-19:
I'm trying to upgrade using Luci "Attended Sysupgrade" tab, but get the error:
[image]

My firmware image is searchable on the build server:
https://4pda.to/pages/go/?u=https%3A%2F%2Fchef.libremesh.org%2F%3Fversion%3D21.02.0-rc3%26target%3Dbcm47xx%252Fmips74k%26id%3Dnetgear_wnr3500l-v1&e=107480544

Caveman8080, 2021-06-29:
If you are using OpenWRT 19.07 or earlier, you might not be able to use attended sysupgrade with https://chef.libremesh.org/ . I believe there was a change to the code and now the website is working with OpenWRT 21.02 a later.

aparcar, 2021-06-29:
I'm in the process of backporting the required changes, it takes a bit more time then expected but I'll try to be done once 21.02 is out.

bam80, 2021-06-29:
Thank you for the update guys. Waiting eagerly, but take your time of course.

aparcar, 2021-07-05:
Fixed for 19.07, I just merged the commit into luci.git. Please give it some 24 hours and install the latest version of luci-app-attendedsysupgrade. Once done it should work just fine.

bam80, 2021-07-05:
Thank you very much! I'll try and report back here in case of any problems.

aparcar, 2021-08-05:
I'd close this unless you have any further issues with the server?

closed 2021-08-29 by aparcar

bam80, 2021-11-07:
> Fixed for 19.07, I just merged the commit into luci.git. Please give it some 24 hours and install the latest version of luci-app-attendedsysupgrade. Once done it should work just fine.

@aparcar still doesn't work here but the error is a bit different now:

> Error connecting to upgrade server
> 
> Could not reach API at "https://chef.libremesh.org/". Please try again later.
> 
> TypeError: can't convert undefined to object

https://github.com/openwrt/luci/issues/4280