openwrt/asu #285: Customized image again down

locussss, 2021-08-05:
Hi!

As subject says: again 404 with chef.libremesh.org/api/build

aparcar, 2021-08-05:
Please try asu.aparcar.org

locussss, 2021-08-05:
Can I substitute https://chef.libremesh.org <https://chef.libremesh.org/> with this address https://asu.aparcar.org in luci-app-attendedsysupgrade?

Regards,

Hans

aparcar, 2021-08-05:
> Can I substitute https://chef.libremesh.org <https://chef.libremesh.org/> with this address https://asu.aparcar.org in luci-app-attendedsysupgrade?
> 
> Regards,
> 
> Hans
> 
> > 
> > 
> > Please try asu.aparcar.org

Correct, please share your results

locussss, 2021-08-05:
Okay, changed address in new address, got my popup with installed packages, adjusted Board Name/Profile with rpi-4 and clicked Request Sysupgrade.
After a couple of minutes the build was ready but download from server to browser didn’t finish.

I went to https://asu.aparcar.org <https://asu.aparcar.org/> to the store/SNAPSHOT/bcm27xx/bcm2711/rpi-4 and I saw map with random number at my build time.

From there I downloaded the sysupgrade image and manually upgraded.

So it was okay but not yet completely automated.

Hans

aparcar, 2021-08-05:
CORS should be fixed, please try again and if it works close this.

closed 2021-08-21 by aparcar