openwrt/asu #289: Log Files for ImageBuilder / ASU?

richb-hanover, 2021-08-14:
What kind of information does the ImageBuilder collect?

One big question we asked at the OpenWrt meeting in Hamburg (June 2019) was whether we should collect statistics about devices, packages, etc. We wanted actual data to inform decisions about popular devices, Flash/RAM sizes, versions, packages, etc. so we could decide how to spend time/developer resources/etc. 

We kicked around the idea of having a device "phone home" and tell some central OpenWrt server its device type, firmware version, packages, and maybe other info. But we were extremely reluctant to do this without an extremely detailed warning. And, in fact, that concern stalled the project.

**But...** the *luci-app-attendedsysupgrade* package has to send in just about all that information in order to request the proper build. So...

* Does it make sense to log the information received by the ImageBuilder?
* How do we talk about any privacy concerns?
* How might we make the logged information available to admin's? To the general public?

Thanks.

aparcar, 2021-08-14:
Good call, maybe @dangowrt can you the stage.

> Does it make sense to log the information received by the ImageBuilder?

We could add some counters to see what images are build, most interesting the `profile` aka `device`
 
> How do we talk about any privacy concerns?

No IPs are logged (except if you try to DDOS the server your IP is logged in a blocklist, nothing else is private data. You're not able to upload private data but only package combinations.

> How might we make the logged information available to admin's? To the general public?

Surely make things public, why not?

richb-hanover, 2021-08-14:
Glad we're on the same page.

I think it's very interesting to log the packages that people request as well as their device. Today, we're left guessing about every aspect of our "customer base". We don't really know what devices they use, the distribution of RAM/Flash, or what packages are important. 

Having statistics about these may not change what we include, but it certainly helps us decide that certain devices/architectures are falling out of favor, and could be de-emphasized in future releases. For example, I think it would have been interesting to know when 4/32 devices began to be less prevalent. I suspect (but have no way of knowing) that we continued to give those devices "Official OpenWrt Support" with stable builds for 6 or 12 months longer than was justified by the effort.

I think we would want the same info for packages: If three out of two  ( :-) images request XYZ package, then we can begin to think about including it in the default build. (It certainly increases the importance of testing that package in the RCx builds.)

But we have some time to decide this. Thanks again.

aparcar, 2021-08-28:
Please see a full list of possible stats [here](https://asu.aparcar.org/ui/)
[image]

Looks like there where about 100 images build in a day, cool.

I'll look into storing these and reset counters every month.

aparcar, 2021-08-28:
Here is an example of popular packages:

https://asu.aparcar.org/api/v1/stats/packages/SNAPSHOT

[image]

The Chef server will become a work server at some point, please use the *new* ASU server at https://asu.aparcar.org for all actual things.

aparcar, 2021-08-29:
Now we would need someone who renders the data via diagrams...

richb-hanover, 2021-08-29:
Cool! I really like the Swagger query tool. Horsing around with it, I had a couple thoughts/questions:

* Is this live data? Is it querying the real server? That's doubly cool!
* Looking at https://asu.aparcar.org/api/v1/stats/images, I immediately wondered, "14 images... since when?" Perhaps the JSON objects that have counts could have a "Since:" property, listing the time it was zeroed out. (That works well with monthly archiving, too - the data self-identifying...
* Might there also be value in a "query" property that specifies the query used? ie `"query": "/v1/stats/packages/SNAPSHOT"`

Thanks again.

richb-hanover, 2021-08-29:
> Now we would need someone who renders the data via diagrams...

And how would you want to visualize the data? What interrelationships would you want to see?  Thanks again for all this great work.

aparcar, 2021-08-29:
> Is this live data? Is it querying the real server? That's doubly cool!

Yes, I might add some caching but right now it's live.

> Looking at https://asu.aparcar.org/api/v1/stats/images, I immediately wondered, "14 images... since when?" Perhaps the JSON objects that have counts could have a "Since:" property, listing the time it was zeroed out. (That works well with monthly archiving, too - the data becomes self-identifying...)

Fair, I just did a database reset therefore the numbers are misleading. I'll work on a monthly rotation. 

> And how would you want to visualize the data? What interrelationships would you want to see? Thanks again for all this great work.

We could work on something similar to this: https://data.syncthing.net/

richb-hanover, 2021-08-29:
> > Is this live data? Is it querying the real server? That's doubly cool!
> 
> Yes, I might add some caching but right now it's live.

That's great! (I was asking whether it was "actual data" instead of test data.) It's real! Cool!
 
> Fair, I just did a database reset therefore the numbers are misleading. I'll work on a monthly rotation.

For the moment, I'm not worried about the startup (it's still in testing) or monthly rotation (although that would probably be interesting.)

I'm asking a slightly different question: A year from now, we'll have a ton of these files (well, probably twelve of each kind :-). Their filename might give us information about their time period, but I was suggesting that we add a JSON property that indicates the date/time so that a reader could get the (start) time of the observations simply by inspecting the file. And that data travels with the rest of the info in the JSON object, so it can't get separated.(That's what I mean about "self-identifying")

> > Might there also be value in a "query" property that specifies the query used? ie "query": "/v1/stats/packages/SNAPSHOT" Again, more self-identification...
> 
> Sure that's possible but I don't see the value of this right now?

It's the same problem. I could wind up with a dozen of these, multiplied by the number of profiles, etc. It's no fun to struggle to determine, "Hmmm... I think that's a v2 query of the profiles..." It seems trivial to include the `query` as another property of the JSON object...

> > And how would you want to visualize the data? What interrelationships would you want to see? Thanks again for all this great work.
> 
> We could work on something similar to this: https://data.syncthing.net/

That's really fascinating! I like it!

aparcar, 2021-09-02:
So now that 21.02 is semi available it's possible to track it's raising to the top :)

[image]

aparcar, 2021-09-24:
I think this can be closed sine we got all the data now available. If you come up with more data to be added, open separate issues.

closed 2021-09-24 by aparcar

aparcar, 2021-09-25:
Added a status overview here https://sysupgrade.openwrt.org/stats

richb-hanover, 2021-09-25:
Sweet! It might help to indicate both the start-time of data collection (i.e., "Builds since: ..." which shows the time the stats were zero'd out) and also the current time, so that people can copy/paste the stats to keep a record of "now" - when they were displayed.

This is so cool - 895 Snapshot builds! Woo hoo!