openwrt/asu #292: Raspberry pi 4b profile error with attendedsysupgrade

locussss, 2021-08-16:
Hi!

Attended sysupgrade through luci is super!

I only have one question (bug?).

When in luci-app-attendedsysupgrade I search for sysupgrade I get the popup dialog which shows me the installed packages with the Board Name/Profile "raspberrypi,4-model-b". This is already pre-filled for me, see screenshot.
[image]

I click request sysupgrade and then I get this error, see screenshot.
[image]

If I replace in Board name/Profile "raspberrypi,4-model-b" with "rpi-4" it runs okay and builds an image.
Why can't it be saved with the correct profile?
I have to change everytime into "rpi-4".

wulfy23, 2021-08-17:
i'm also seeing unsupported profile rpi-2 on the rpi-4

aparcar, 2021-08-17:
Fixed via https://github.com/aparcar/asu/commit/d8ca0f303c01fa1390b373b740789683af102a94 please try again on https://asu.aparcar.org.

locussss, 2021-08-18:
I can confirm it is working now.

Thanks!

closed 2021-08-18 by locussss