openwrt/asu #293: No ASU support in openwrt firmware selector anymore?

egon0, 2021-08-23:
just started to setup a asu server with openwrt firmware selector for internal use and stumbled across this commit:

https://gitlab.com/openwrt/web/firmware-selector-openwrt-org/-/commit/6abd5536f00600be8af68b3b9377703d63d93594

are there any plans on re-enable asu support for the openwrt firmware selector or do the two projects part ways?

aparcar, 2021-08-23:
Ideally the selector becomes the new face of the OpenWrt download page, to do so it should be robust and not contain any half backed features. You can still use my forks `asu-wip` branch, which currently also runs at asu.aparcar.org.

I'm talking with people about modifying the [react wizard](https://github.com/sudhanshu16/openwrt-firmware-selector) to become a frontend for ASU. Since I'm not a JS dev it'll take a bit of time. If you know people interested in that work, please let me know.

egon0, 2021-08-23:
that's a little bit sad (for me atleast) - i really loved the joint effort to have the same ui/frontend for vanilla openwrt and for some more specific images for some edge-cases. now the frontend effortd divert again. currently i have no frontend/js devs with reactjs knowledge in my bubble sadly.

egon0, 2021-08-23:
just played around with the react wizzard. seems to work fine (at least for me :smile:) - as you answered my initial question and gave me some directions to alternative frontends, i'll close this ticket for now.

closed 2021-08-23 by egon0