openwrt/asu #301: Not quite ready for prime-time

thess, 2021-09-24:
OK - I admit that I just thought installing auc, etc. and requesting a new image from my router would work. If it got bricked, I can recover so I am willing to be a guinea pig. Here''s the error output from my first attempted use:

```
Error building the sysupgrade
Unsupported package(s): collectd-mod-rrdtool, irqbalance, coreutils, collectd-mod-interface, firewall, hostip, rrdtool1, coreutils-sort, collectd-mod-memory, librrd1, collectd-mod-iwinfo, collectd, collectd-mod-network, collectd-mod-cpu, collectd-mod-load

Please report the error message and request

Request to server:
{
    "board_name": "netgear,r7800",
    "target": "ipq806x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "adblock",
        "ath10k-board-qca9984",
        "ath10k-firmware-qca9984",
        "auc",
        "base-files",
        "block-mount",
        "blockd",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "coreutils",
        "coreutils-sort",
        "ddns-scripts",
        "ddns-scripts-services",
        "dnscrypt-proxy2",
        "dnsmasq-dhcpv6",
        "dosfstools",
        "drill",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "hostip",
        "ip6tables",
        "iptables",
        "irqbalance",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-sha256",
        "kmod-fs-autofs4",
        "kmod-fs-ext4",
        "kmod-fs-msdos",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-input-core",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc16",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nls-base",
        "kmod-nls-cp1250",
        "kmod-nls-cp437",
        "kmod-nls-cp850",
        "kmod-nls-iso8859-1",
        "kmod-nls-iso8859-15",
        "kmod-nls-iso8859-2",
        "kmod-nls-utf8",
        "kmod-phy-qcom-ipq806x-usb",
        "kmod-scsi-core",
        "kmod-sound-core",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ehci",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb-storage",
        "kmod-usb-xhci-hcd",
        "kmod-usb2",
        "kmod-usb3",
        "libblobmsg-json",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "libldns",
        "libltdl",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libncurses",
        "libnl-tiny",
        "libopenssl",
        "libpcap",
        "librrd1",
        "libsodium",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libusb-1.0",
        "libustream-openssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-httpprotoutils",
        "luci-lib-ip",
        "luci-lib-ipkg",
        "luci-lib-iptparser",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "luci-theme-material",
        "luci-theme-openwrt",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "procd",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rrdtool1",
        "swconfig",
        "tcpdump-mini",
        "terminfo",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wget-ssl",
        "wireless-regdb",
        "wireless-tools",
        "wpad-basic",
        "zile",
        "zlib"
    ],
    ".anonymous": false,
    ".name": "request",
    ".type": "request",
    ".index": 0
}
```

aparcar, 2021-09-24:
Hi @thess thanks for testing!

The mentioned packages are [currently missing](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/) due to some [libtool fallout](https://github.com/openwrt/openwrt/commit/96e05e2e36fc3d0a893702a7c445aac0348320a4). Please wait a day until those missing packages are build again and try again.

thess, 2021-09-24:
Good, I'll try again this weekend. I also realized that my customized additional options on packages like *busybox*, etc. will be lost by this procedure. Caveat User.

thess, 2021-09-24:
Maybe the error message should allude to the problem (internal) rather than just calling out the packages as unsupported.

aparcar, 2021-09-24:
I'm thinking of some better error logging to realize such issues faster. Will think about something.

For special busybox versions, with SELinux I added a special variant, for minor changes I can't think of a clean solution...

thess, 2021-09-30:
Worked for me over the weekend -- closing

closed 2021-09-30 by thess