openwrt/asu #302: Web landing page could use some work

thess, 2021-09-30:
At a minimum, some branding and better looking layout.
Current page is pretty spartan.

aparcar, 2021-10-01:
If someone steps up and beautifies things I'm happy to merge things, however I'm not very keen to do any web stuff myself.

richb-hanover, 2021-10-08:
I'm working on beautifying the page. I've got some styling working, but I'm not sure how to introduce some test data into either the Overview or the Stats page. (I've followed the [Development steps in the README](https://github.com/aparcar/asu#development) but there doesn't seem to be any data. However, the `tests/conftest.py` looks interesting, but I don't know what to do with it...)

aparcar, 2021-10-09:
If you install the development requirements they are already included