openwrt/asu #310: Add 21.02.1 to ASU/sysupgrade.openwrt.org

richb-hanover, 2021-10-30:
It's probably on your list, but the luci-app-attendedsysupgrade does not yet find the 21.02.1 build. (Tested with 21.02.0 on Netgear WNDR3800 and D-Link DIR-878)

But I'm very excited to try it out! :-) Thanks!

aparcar, 2021-11-02:
Added

```shell
BusyBox v1.33.1 (2021-10-19 07:09:12 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.0, r16279-5cc0535800
 -----------------------------------------------------
root@gateway:~# auc
auc (0.2.4-1)
Server:    http://sysupgrade.openwrt.org
Running:   21.02.0 r16279-5cc0535800 on ath79/generic (tplink,archer-a7-v5)
Available: 21.02.1 r16325-88151b8303
Requesting package lists...
 kmod-usb-core: 5.4.143-1 -> 5.4.154-1
--- %< ---
 kmod-slhc: 5.4.143-1 -> 5.4.154-1
 kmod-cfg80211: 5.4.143+5.10.42-1-1 -> 5.4.154+5.10.68-1-1
 kmod-ipt-nat: 5.4.143-1 -> 5.4.154-1
Are you sure you want to continue the upgrade process? [N/y] 
```

closed 2021-11-02 by aparcar

richb-hanover, 2021-11-02:
I just have to say what an immense improvement `luci-app-attendedsysupgrade` makes. I just upgraded three routers in three clicks. (Well, six clicks - I had to click OK for each...)

It changes a [**slog** - a focus-intensive process that involves very little thought](https://blog.danslimmon.com/2019/07/15/do-nothing-scripting-the-key-to-gradual-automation/) - into a pleasant single-button step. It has always (after a half-dozen tests...) left me with a perfectly working router.

Thank you.

PS For others who want to try this at home, make sure the configuration points to `sysupgrade.openwrt.org` or `asu.aparcar.org`