openwrt/asu #313: How to reset the queued request in build?

kiddin9, 2022-01-16:
Hi,
I build my owner server: https://op.supes.top/api/overview ,But it's seem not working properly. I got this error all the time.

[image]
[image]

This is my config.py
```
from pathlib import Path

# disable test and debug features
TESTING = False
DEBUG = True

# where to find the ImageBuildes
UPSTREAM_URL = "https://downloads.openwrt.org"

# where to store created images
STORE_PATH = Path.cwd() / "public/store/"

# where to store JSON files
JSON_PATH = Path.cwd() / "public/json/v1/"

MAPPING_ABI = {"libubus20191227": "libubus"}

# supported versions
BRANCHES = {
    "SNAPSHOT": {
        "name": "SNAPSHOT",
        "updates": "dev",
        "enabled": True,
        "snapshot": True,
        "versions": ["SNAPSHOT"],
        "git_branch": "master",
        "path": "snapshots",
        "path_packages": "snapshots/packages",
        "pubkey": "RWS1BD5w+adc3j2Hqg9+b66CvLR7NlHbsj7wjNVj0XGt/othDgIAOJS+",
        "repos": ["base", "packages", "luci", "routing", "telephony"],
        "extra_repos": {},
        "extra_keys": [],
        "targets": {"x86/64":"x86_64"},
    },
}
```

I have no idea what wrong, I will very appreciated if anyone can help me out. Thanks.

aparcar, 2022-01-16:
Hi, thanks for testing this service! Is it possible that you don't run any worker? Please try to run a worker via the [systemd service](https://github.com/aparcar/asu/blob/master/misc/rqworker.service) or within e.g. a `tmux` session by running `rq worker`.

kiddin9, 2022-01-16:
Sorry, I ain't got no sense in python, Can't you tell me how to run a worker and janitor update via systemd service? Thanks.

kiddin9, 2022-01-17:
Hi, I've changed the UPSTREAM_URL to my own downloads server: https://op.supes.top/releases/, then I got a new error:

[image]

 my config.py
```
from pathlib import Path

# disable test and debug features
TESTING = False
DEBUG = True

# where to find the ImageBuildes
UPSTREAM_URL = "https://op.supes.top"
# UPSTREAM_URL = "https://mirrors.cloud.tencent.com/openwrt"

# where to store created images
STORE_PATH = "/www/wwwroot/op.supes.top/cache/store/"

# where to store JSON files
JSON_PATH = Path.cwd() / "public/json/v1/"

MAPPING_ABI = {"libubus20191227": "libubus"}

# ALLOW_DEFAULTS = True

# supported versions
BRANCHES = {
    "21.02": {
        "name": "21.02",
        "updates": "features",
        "enabled": True,
        "snapshot": False,
        "versions": ["21.02.1"
        ],
        "git_branch": "openwrt-21.02",
        "path": "releases/{version}",
        "path_packages": "releases/21.02.1/packages",
        "pubkey": "",
        "repos": ["base", "packages", "luci", "routing"],
        "targets": {"x86/64":"x86_64"},
    },
}
```

Thank you for your patience.

aparcar, 2022-01-17:
Currently ASU determines the available targets with a poorly documented JSON file. It's available upstream but not produced by the build system. I tried to establish a JSON file upstream but no outcome just yet.

Specifically ASU will look for a file called `?json-targets` in the `/targets/` folder of a release, below I linked an example:

https://downloads.openwrt.org/snapshots/targets/?json-targets

This file contains a list (array) or available targets. For you the content of the file would simply be `[ "x86/64" ]`.

aparcar, 2022-01-17:
I know that this approach is far from pretty, however there is currently no automated way to do this. If ASU would parse HTML instead, it would crash whenever someone uses a different HTML theme.

kiddin9, 2022-01-17:
Yes, I noticed this issue, this is why I set `"targets": {"x86/64":"x86_64"}` in my config.py and reverted https://github.com/aparcar/asu/blob/65c551c240ae70991f8d915a3e61e143ee8e4b20/asu/janitor.py#L86 to `targets = branch["targets"].keys()`

Is there got anything else I need take care of?  Thanks

kiddin9, 2022-01-17:
Sorry, I set wrong target value, I'll try "targets": [ "x86/64" ] and feedback. Thanks.

kiddin9, 2022-01-17:
It seem I didn't wrong, the branch["targets"].keys() just x86/64,  Is there got anything else I need take care of? Thanks.

aparcar, 2022-01-17:
I just tested and it works fine with your server, see the log below:

```
Loading /Users/user/src/openwrt/bsu/config.py
[2022-01-17 14:18:41,343] INFO in janitor: Init ASU janitor
[2022-01-17 14:18:41,343] INFO in janitor: Update 21.02
[2022-01-17 14:18:41,344] INFO in janitor: Update 21.02/21.02.1
[2022-01-17 14:18:41,344] INFO in janitor: 21.02.1/x86/64: Update packages
[2022-01-17 14:18:43,658] INFO in janitor: x86/64: Add ABI mapping libgcc1 -> libgcc
[2022-01-17 14:18:43,659] INFO in janitor: x86/64: Add ABI mapping libip4tc2 -> libip4tc
[2022-01-17 14:18:43,659] INFO in janitor: x86/64: Add ABI mapping libip6tc2 -> libip6tc
[2022-01-17 14:18:43,659] INFO in janitor: x86/64: Add ABI mapping libstdcpp6 -> libstdcpp
[2022-01-17 14:18:43,660] INFO in janitor: x86/64: Add ABI mapping libxtables12 -> libxtables
[2022-01-17 14:18:43,660] DEBUG in janitor: 21.02.1/x86/64: Found 946
[2022-01-17 14:18:43,665] INFO in janitor: 21.02.1: found 946 packages
[2022-01-17 14:18:43,667] INFO in janitor: 21.02.1/x86/64: Update profiles
[2022-01-17 14:18:46,570] INFO in janitor: 21.02.1/x86/64: Found 1 profiles
[2022-01-17 14:18:46,573] INFO in janitor: Update 21.02/x86_64
[2022-01-17 14:18:48,303] INFO in janitor: base: Add ABI mapping libblkid1 -> libblkid
[2022-01-17 14:18:48,304] INFO in janitor: base: Add ABI mapping libblobmsg-json20210516 -> libblobmsg-json
[2022-01-17 14:18:48,304] INFO in janitor: base: Add ABI mapping libbpf0 -> libbpf
[2022-01-17 14:18:48,305] INFO in janitor: base: Add ABI mapping libcomerr0 -> libcomerr
[2022-01-17 14:18:48,305] INFO in janitor: base: Add ABI mapping libelf1 -> libelf
[2022-01-17 14:18:48,305] INFO in janitor: base: Add ABI mapping libevent2-7 -> libevent2
[2022-01-17 14:18:48,306] INFO in janitor: base: Add ABI mapping libext2fs2 -> libext2fs
[2022-01-17 14:18:48,306] INFO in janitor: base: Add ABI mapping libf2fs6 -> libf2fs
[2022-01-17 14:18:48,307] INFO in janitor: base: Add ABI mapping libgmp10 -> libgmp
[2022-01-17 14:18:48,307] INFO in janitor: base: Add ABI mapping libipset13 -> libipset
[2022-01-17 14:18:48,307] INFO in janitor: base: Add ABI mapping libiwinfo20210430 -> libiwinfo
[2022-01-17 14:18:48,308] INFO in janitor: base: Add ABI mapping libjson-c5 -> libjson-c
[2022-01-17 14:18:48,308] INFO in janitor: base: Add ABI mapping libjson-script20210516 -> libjson-script
[2022-01-17 14:18:48,308] INFO in janitor: base: Add ABI mapping liblua5.1.5 -> liblua
[2022-01-17 14:18:48,309] INFO in janitor: base: Add ABI mapping libmbedtls12 -> libmbedtls
[2022-01-17 14:18:48,309] INFO in janitor: base: Add ABI mapping libmnl0 -> libmnl
[2022-01-17 14:18:48,309] INFO in janitor: base: Add ABI mapping libncurses6 -> libncurses
[2022-01-17 14:18:48,310] INFO in janitor: base: Add ABI mapping libnetfilter-conntrack3 -> libnetfilter-conntrack
[2022-01-17 14:18:48,310] INFO in janitor: base: Add ABI mapping libnettle8 -> libnettle
[2022-01-17 14:18:48,310] INFO in janitor: base: Add ABI mapping libnfnetlink0 -> libnfnetlink
[2022-01-17 14:18:48,310] INFO in janitor: base: Add ABI mapping libnl-core200 -> libnl-core
[2022-01-17 14:18:48,311] INFO in janitor: base: Add ABI mapping libnl-genl200 -> libnl-genl
[2022-01-17 14:18:48,311] INFO in janitor: base: Add ABI mapping libnl-tiny1 -> libnl-tiny
[2022-01-17 14:18:48,311] INFO in janitor: base: Add ABI mapping libopenssl1.1 -> libopenssl
[2022-01-17 14:18:48,312] INFO in janitor: base: Add ABI mapping libreadline8 -> libreadline
[2022-01-17 14:18:48,312] INFO in janitor: base: Add ABI mapping libsmartcols1 -> libsmartcols
[2022-01-17 14:18:48,312] INFO in janitor: base: Add ABI mapping libss2 -> libss
[2022-01-17 14:18:48,313] INFO in janitor: base: Add ABI mapping libsysfs2 -> libsysfs
[2022-01-17 14:18:48,313] INFO in janitor: base: Add ABI mapping libubox20210516 -> libubox
[2022-01-17 14:18:48,313] INFO in janitor: base: Add ABI mapping libubus20210630 -> libubus
[2022-01-17 14:18:48,313] INFO in janitor: base: Add ABI mapping libuci20130104 -> libuci
[2022-01-17 14:18:48,314] INFO in janitor: base: Add ABI mapping libuclient20201210 -> libuclient
[2022-01-17 14:18:48,314] INFO in janitor: base: Add ABI mapping libustream-wolfssl20201210 -> libustream-wolfssl
[2022-01-17 14:18:48,314] INFO in janitor: base: Add ABI mapping libuuid1 -> libuuid
[2022-01-17 14:18:48,315] INFO in janitor: base: Add ABI mapping libwolfssl4.8.1.66253b90 -> libwolfssl
[2022-01-17 14:18:48,315] DEBUG in janitor: 21.02/x86_64/base: Found 131 packages
[2022-01-17 14:18:49,332] INFO in janitor: packages: Add ABI mapping libcurl4 -> libcurl
[2022-01-17 14:18:49,332] INFO in janitor: packages: Add ABI mapping libnghttp2-14 -> libnghttp2
[2022-01-17 14:18:49,333] INFO in janitor: packages: Add ABI mapping libsensors5 -> libsensors
[2022-01-17 14:18:49,333] DEBUG in janitor: 21.02/x86_64/packages: Found 81 packages
[2022-01-17 14:18:50,180] INFO in janitor: luci: Add ABI mapping liblucihttp0 -> liblucihttp
[2022-01-17 14:18:50,180] DEBUG in janitor: 21.02/x86_64/luci: Found 40 packages
[2022-01-17 14:18:50,834] DEBUG in janitor: 21.02/x86_64/routing: Found 1 packages
[2022-01-17 14:18:50,838] INFO in janitor: x86_64: found 253 packages
[2022-01-17 14:18:50,839] INFO in janitor: Next reload in 10 minutes
```

I used your server as upstream and just manually set `targets = ["x86/64"]` in `janitor.py`.

kiddin9, 2022-01-17:
My janitor.py works fine as well, but worker report HTTP Error 404: Not Found all the time.  Could you test this command for me? Thanks for your patience.

kiddin9, 2022-01-17:
FYI, janitor.py don't log any debug and info logs even though I set DEBUG = True in config.py.

kiddin9, 2022-01-23:
Sorry for the late feedback, I have been pretty hectic for a while.
I finally pulled it off, My server: https://op.supes.top is working fantabulous,
I deployed the website client over https://github.com/sudhanshu16/openwrt-firmware-selector, But it can't check and show available packages autocomplete list, Cant't edit custom package list with text, That is embarrassed, The https://gitlab.com/openwrt/web/firmware-selector-openwrt-org you commended have been droping ASU support. I was wondering if you can share your ASU modification version with me.

Thanks for your incredible working!

aparcar, 2022-01-26:
> I was wondering if you willing to share your ASU modification version's OFS with me.

I'll have to rebase it and make it work with the latest modifications of ASU. I've never been fully satisfied with the integration I did so I'm somewhat looking for a javascript dev willing to implement it. Do you know anyone?

aparcar, 2022-03-09:
Here is the currently deployed branch/source, however I'd prefer using something like preact or so in the future:

https://gitlab.com/aparcar/firmware-selector-openwrt-org/-/commit/15cc718ed3c4cd2266357597338b7d64a30f0081

aparcar, 2022-06-07:
I gave the integration another go here, closing this as the initial issue seems resolved

https://gitlab.com/openwrt/web/firmware-selector-openwrt-org/-/merge_requests/100

closed 2022-06-07 by aparcar