openwrt/asu #315: Save a copy of the image

chuacw, 2022-01-27:
Provide a way to save a copy of the image before sysupgrade.

Currently, there's no way to save it.

aparcar, 2022-01-27:
Using `arc` shows you the full download URL, using the LuCI app you can click on the filename to download it.

chuacw, 2022-01-27:
I checked using auc --arc but I don't see the full download URL.
Is there some flag I need to set to see it?

[image]

As for the LuCI app, I don't see where I can see the filename either.
Are we talking about the same Unattended Sysupgrade app?

[image]

dangowrt, 2022-01-27:
I've added a option `-n` to `auc` to allow users to only request the image but not download or sysupgrade anything (dry-run).
https://github.com/openwrt/packages/commit/19c135685f7b9eefbf75b5a96b780f4fa2b46f19

chuacw, 2022-01-28:
There's no blue text.

[image]

chuacw, 2022-01-28:
@dangowrt I would love for the app to have the ability to download, and show where it downloaded the image to, so that the user can save a copy of the image, prompt to continue, before starting the upgrade.

Currently, as a user, there's no way from the CLI to make a copy of the image.

dangowrt, 2022-01-28:
@chuacw I've just added that feature to `auc`, you can now call `auc -n` to request the image and show the download URL, but not carry out the download or sysupgrade.
https://github.com/aparcar/asu/issues/315#issuecomment-1023470485
https://github.com/openwrt/packages/commit/19c135685f7b9eefbf75b5a96b780f4fa2b46f19

chuacw, 2022-01-28:
Hi @dangowrt yes, before I posted my previous response, I've already seen your response that you've added "-n".
My request is to save the image, not just provide the download URL.

dangowrt, 2022-01-28:
Once you got the URL, you can use `wget` (aka. `uclient-fetch`) to download it to `/tmp`...?

chuacw, 2022-01-28:
When auc is called to perform the upgrade, it already downloads the image to the system.
Display where it downloaded the image to and pausing at this point (with a prompt) to allow the user to copy out the image from the system.

chuacw, 2022-01-28:
auc doesn't show the URL at all.

[image]

[image]

dangowrt, 2022-01-28:
It has just been committed to https://github.com/openwrt/packages/ repository (which, btw, would be the right place to file issues for `auc`, this repository is about the server side only). This means you still have to update the package on your device, and you can do that once the buildbot has created new binaries (that may take up to 2 days). So unsurprisingly, you can't enjoy that new feature just yet, just because I've added it in git it won't instantly be available in your device...

closed 2022-01-28 by dangowrt