openwrt/asu #319: Different profile built than requested

n-thumann, 2022-02-27:
Hi :v:

I've tried to use the attended sysupgrade to upgrade my router, but it seems like the wrong profile has been built.

Steps to reproduce:
`curl 'https://sysupgrade.openwrt.org/api/v1/build' -X POST -H 'Content-Type: application/json' --data-raw '{"profile":"glinet,gl-ar750s-nor-nand","target":"ath79/nand","version":"21.02.2","packages":["dropbear"],"diff_packages":true}` (only one package, just as an example. Notice the profile being `gl-ar750s-nor-nand`)

Response (truncated in some places):
```
{
  "arch_packages": "mips_24kc",
  "bin_dir": "dfdfe2cde035a4f942c2da123459675d",
  "build_at": "2022-02-27T11:26:10.689907Z",
  "build_cmd": [
    "make",
    "image",
    "PROFILE=glinet_gl-ar750s-nor",
    "PACKAGES=-odhcp6c -libustream-wolfssl -odhcpd-ipv6only -kmod-ipt-offload -urngd -procd -uclient-fetch -uci -iptables -netifd -block-mount -swconfig -ip6tables -wpad-basic-wolfssl -ppp -kmod-gpio-button-hotplug -dnsmasq dropbear -libc -urandom-seed -kmod-ath10k-ct -base-files -firewall -kmod-usb2 -logd -busybox -ppp-mod-pppoe -ca-bundle -ath10k-firmware-qca9887-ct -opkg -uboot-envtools -kmod-ath9k -kmod-usb-storage -fstools -libgcc -mtd",
    "EXTRA_IMAGE_NAME=d87a1b97eb31",
    "BIN_DIR=/home/aparcar/asu/public/store/dfdfe2cde035a4f942c2da123459675d"
  ],
  [...]
  "enqueued_at": "2022-02-27T11:26:01.569949Z",
  "id": "glinet_gl-ar750s-nor",
  "image_prefix": "openwrt-21.02.2-d87a1b97eb31-ath79-nand-glinet_gl-ar750s-nor",
  "imagebuilder_status": "building_image",
  "images": [
    {
      "filesystem": "squashfs",
      "name": "openwrt-21.02.2-d87a1b97eb31-ath79-nand-glinet_gl-ar750s-nor-squashfs-sysupgrade.bin",
      "sha256": "2d2cebc45645072174ed1254d09e564bcd042dd280340f856cbc760c9d59778d",
      "type": "sysupgrade"
    }
  ],
  "manifest": {
    "dropbear": "2020.81-2",
    "kernel": "5.4.179-1-f8e3efb079bffea9f52f6412f557cb64",
    "libc": "1.1.24-3",
    "libgcc": "8.4.0-3"
  },
  "metadata_version": 1,
  "request_hash": "dfdfe2cde035a4f942c2da123459675d",
  "source_date_epoch": "1645043350",
  "status": 200,
  "stderr": "Package list missing or not up-to-date, generating it.\n\nBuilding package index...\nDownloading http://downloads.openwrt.org/releases/21.02.2/targets/ath79/nand/packages/Packages.gz [...]",
  "stdout": "Building images for ath79 - GL.iNet GL-AR750S NOR [...]",
  "supported_devices": [
    "glinet,gl-ar750s-nor",
    "gl-ar750s",
    "glinet,gl-ar750s",
    "glinet,gl-ar750s-nor-nand"
  ],
  "target": "ath79/nand",
  "titles": [
    {
      "model": "GL-AR750S",
      "variant": "NOR",
      "vendor": "GL.iNet"
    }
  ],
  "version_code": "r16495-bf0c965af0",
  "version_number": "21.02.2"
}
```
Note that the changed to `glinet_gl-ar750s-nor` and hence the file name being `openwrt-21.02.2-d87a1b97eb31-ath79-nand-glinet_gl-ar750s-nor-squashfs-sysupgrade.bin`. It looks like the profile has been changed from `gl-ar750s-nor-nand` to `gl-ar750s-nor` before passing it to the builder.

When I run the exact same request against my local asu instance, the profile is not changed at any point and `make image PROFILE=glinet_gl-ar750s-nor-nand [...]` is run by the worker as expected.

n-thumann, 2022-02-27:
I assume this is due to https://downloads.openwrt.org/releases/21.02.2/targets/ath79/nand/profiles.json containing `"glinet,gl-ar750s-nor-nand"` in both the `glinet_gl-ar750s-nor` and `glinet_gl-ar750s-nor-nand` profile. Because the `...-nor` one is before the `...-nor-nand` one, it's set as the mapping in Redis.
One can manually work around this temporarily by setting the mapping using `HSET mapping-21.02-21.02.2-ath79/nand glinet,gl-ar750s-nor-nand glinet_gl-ar750s-nor-nand`.

aparcar, 2022-02-27:
Problem seem to be a mapping issue, I have a patch in mind and will try to offer an update in a little bit.

aparcar, 2022-02-27:
Could you please try this patch locally?

```diff
diff --git a/asu/api.py b/asu/api.py
index 99e68ba..90d66d4 100644
--- a/asu/api.py
+++ b/asu/api.py
@@ -239,25 +239,25 @@ def validate_request(req):
         current_app.logger.debug("Use generic profile for {req['target']}")
         req["profile"] = "generic"
     else:
-        mapped_profile = r.hget(
-            f"mapping-{req['branch']}-{req['version']}-{req['target']}",
-            req["profile"],
-        )
-
-        if mapped_profile:
-            req["profile"] = mapped_profile.decode()
-
-        current_app.logger.debug("Profile after mapping " + req["profile"])
-
-        if not r.sismember(
+        if r.sismember(
             f"profiles-{req['branch']}-{req['version']}-{req['target']}",
-            req["profile"],
+            req["profile"].replace(",", "_"),
         ):
-            return (
-                {"detail": f"Unsupported profile: {req['profile']}", "status": 400},
-                400,
+            req["profile"] = req["profile"].replace(",", "_")
+        else:
+            mapped_profile = r.hget(
+                f"mapping-{req['branch']}-{req['version']}-{req['target']}",
+                req["profile"],
             )
 
+            if mapped_profile:
+                req["profile"] = mapped_profile.decode()
+            else:
+                return (
+                    {"detail": f"Unsupported profile: {req['profile']}", "status": 400},
+                    400,
+                )
+
     package_problems = validate_packages(req)
     if package_problems:
         return package_problems
```

n-thumann, 2022-02-27:
> Could you please try this patch locally?

Unfortunately, this doesn't work for me. The log still contains (just like before):
```
asu-server-1     | [2022-02-27 12:47:23,816] DEBUG in api: Profile before mapping glinet,gl-ar750s-nor-nand
asu-server-1     | [2022-02-27 12:47:23,817] DEBUG in api: Profile after mapping glinet_gl-ar750s-nor
```
Strangely enough, I had to manually set run `HSET mapping-21.02-21.02.2-ath79/nand glinet,gl-ar750s-nor-nand glinet_gl-ar750s-nor` to tamper with the key, because now the issue is basically inverted:
Instead of `glinet,gl-ar750s-nor-nand` and `glinet,gl-ar750s-nor` both containing `glinet_gl-ar750s-nor`, they now both contain `glinet_gl-ar750s-nor-nand` 🤔

n-thumann, 2022-02-27:
> Could you please try this patch locally?

```
asu-server-1     | [2022-02-27 16:41:49,104] DEBUG in api: req {'profile': 'glinet,gl-ar750s-nor-nand', 'target': 'ath79/nand', 'version': '21.02.2', 'packages': ['wpad-openssl'], 'diff_packages': True}
asu-server-1     | [2022-02-27 16:41:49,106] DEBUG in api: Profile before mapping glinet,gl-ar750s-nor-nand
asu-server-1     | [2022-02-27 16:41:49,113] DEBUG in api: {'status': 202, 'detail': 'queued', 'queue_position': 0, 'enqueued_at': datetime.datetime(2022, 2, 27, 16, 41, 49, 109160), 'request_hash': 'cc2473225026d5d19dc43d9d3769f8e6'}
asu-worker2-1    | 16:41:49 default: asu.build.build({'profile': 'glinet_gl-ar750s-nor-nand', 'target': 'ath79/nand', 'version':...) (cc2473225026d5d19dc43d9d3769f8e6)
[...]
asu-server-1     | [2022-02-27 16:43:31,240] DEBUG in api: req {'profile': 'glinet,gl-ar750s-nor', 'target': 'ath79/nand', 'version': '21.02.2', 'packages': ['wpad-openssl'], 'diff_packages': True}
asu-server-1     | [2022-02-27 16:43:31,242] DEBUG in api: Profile before mapping glinet,gl-ar750s-nor
asu-worker-1     | 16:43:31 default: asu.build.build({'profile': 'glinet_gl-ar750s-nor', 'target': 'ath79/nand', 'version': '21....) (b8b0f9d5888545e9954042a3f32a3f79)
```
Works like a charm! It now builds the requested variant, no matter if I request `...-nor` or `...-nor-nand` 👍🏻

PR #321 (2022-02-28): api: prefer profile directly over mapped one
Within OpenWrt the profile does not always matches the `board_name`,
which is determined via device tree files. As a workaround ASU
automatically parsed the "supported" devices and mapped incomming
requests to profiles. In some rare cases (#319) the support profile is
actually a different profile than the one installed, in this case a
device that support images with and without nand storage.

To fix this, first check if a profile with the device tree model exists
(by replacing `,` with `_`) and only use the *supported mapping* if it
does not.

Fixes [... cut]
commit b3215757 (2022-02-28): api: prefer profile directly over mapped one

Within OpenWrt the profile does not always matches the `board_name`,
which is determined via device tree files. As a workaround ASU
automatically parsed the "supported" devices and mapped incomming
requests to profiles. In some rare cases (#319) the support profile is
actually a different profile than the one installed, in this case a
device that support images with and without nand storage.

To fix this, first check if a profile with the device tree model exists
(by replacing `,` with `_`) and only use the *supported mapping* if it
does not.

Fixe [... cut]