openwrt/asu #329: Sysupgrade fails for Linksys E8450

chuacw, 2022-03-01:
Error building the sysupgrade

asu.build.PackageSelectionError

Please report the error message and request
Request to server:

```
{
    "board_name": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "blockd",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "ip6tables",
        "iptables",
        "kernel",
        "kmod-fs-exfat",
        "kmod-fs-ext4",
        "kmod-fs-f2fs",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb3",
        "libc",
        "libustream-openssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-sqm",
        "luci-app-upnp",
        "luci-app-watchcat",
        "luci-ssl-openssl",
        "luci-theme-openwrt-2020",
        "miniupnpd-nftables",
        "mtd",
        "nano",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "uhttpd-mod-lua",
        "urandom-seed",
        "urngd",
        "wpad-openssl"
    ],
    ".anonymous": false,
    ".name": "request",
    ".type": "request",
    ".index": 0
}
```

STDERR:

```
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.10.100-1-f2a1ec5d9cc3dd4c7932fbc655c31374/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.10.100-1-f2a1ec5d9cc3dd4c7932fbc655c31374/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubus20210809 for blockd
 * pkg_hash_fetch_best_installation_candidate: Packages for blockd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package blockd.
make[2]: *** [Makefile:168: package_install] Error 255
make[1]: *** [Makefile:133: _call_manifest] Error 2
make: *** [Makefile:252: manifest] Error 2
```

aparcar, 2022-03-01:
I just tested it and without the `blockd` package it works fine. I think this is an upstream problem which is hopefully fixed with the next Buildbot cycle.

chuacw, 2022-03-01:
Thanks for the update, @aparcar . I've removed the blockd package.

closed 2022-03-01 by chuacw