openwrt/asu #342: linksys e8450-ubi: Error building the firmware image

mutantmell, 2022-03-24:
When attempting to do an attended sysupgrade, I get the following error:

> Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19138-4ecf8346c0",
    "advanced_mode": "0",
    "app_version": "git-22.068.37885-65266c4",
    "branch": "SNAPSHOT",
    "efi": null,
    "rootfs_type": "squashfs",
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "blockd",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "kernel",
        "kmod-fs-exfat",
        "kmod-fs-ext4",
        "kmod-fs-f2fs",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb3",
        "libc",
        "libustream-openssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-ssl-openssl",
        "luci-theme-openwrt-2020",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-openssl"
    ],
    "diff_packages": true
}
```

mutantmell, 2022-03-24:
Trying this with `https://asu.aparcar.org` gives me a different error:

> Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

```json
{
    "url": "https://asu.aparcar.org",
    "revision": "r19138-4ecf8346c0",
    "advanced_mode": "0",
    "app_version": "git-22.068.37885-65266c4",
    "branch": "SNAPSHOT",
    "efi": null,
    "rootfs_type": "squashfs",
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "blockd",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "kernel",
        "kmod-fs-exfat",
        "kmod-fs-ext4",
        "kmod-fs-f2fs",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb3",
        "libc",
        "libustream-openssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-ssl-openssl",
        "luci-theme-openwrt-2020",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-openssl"
[... 38 lines cut ...]
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://feed.libremesh.org/master/Packages.gz
Updated list of available packages in /home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/lime-packages
Downloading https://feed.libremesh.org/master/Packages.sig
Signature check passed.
Downloading https://feed.libremesh.org/profiles/Packages.gz
Updated list of available packages in /home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/debian/metrics/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/lime-profiles
Downloading https://feed.libremesh.org/profiles/Packages.sig
Signature check passed.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libustream-openssl for luci-ssl-openssl
 * pkg_hash_check_unresolved: cannot find dependency openssl-util for luci-ssl-openssl
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-ssl-openssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-ssl-openssl.
 * opkg_install_cmd: Cannot install package libustream-openssl.
 * opkg_install_cmd: Cannot install package wpad-openssl.
make[2]: *** [Makefile:168: package_install] Error 255
make[1]: *** [Makefile:133: _call_manifest] Error 2
make: *** [Makefile:252: manifest] Error 2
```

aparcar, 2022-03-27:
I think there was an issue with OpenSSL in snapshots. Could you please try again?

mutantmell, 2022-03-27:
Can confirm that it works now - cheers!

closed 2022-03-27 by aparcar